srobo/j5 #6: [QUESTION] Are BackendGroups extensible?

kierdavis, 2019-01-24:
From my understanding of how j5 is currently structured (please correct me if I'm wrong), it looks like a `BackendGroup` must have knowledge of every type of board that could provide. It also doesn't look like `HardwareBackendGroup` (and presumably other types of `BackendGroup` too, once they are defined) is competition-specific, so a `HardwareBackendGroup` therefore needs to have every board used in every competition hardcoded into it. Additionally, I'm not certain but it sounds like this structure would restrict the ability for end users to define their own boards using the j5 API.

If this is indeed true, my proposal would be for `BackendGroup`s to be initially empty, and to have each `Backend` register itself into its corresponding `BackendGroup` at the time of class definition (e.g. using [`__init_subclass__`](https://docs.python.org/3/reference/datamodel.html#object.__init_subclass__).

trickeydan, 2019-01-25:
I like this idea and can definitely see the benefits that it would offer. When I get a chance I will have a think to consider how this could be done with the current project structure, or what changes would need to be made.

trickeydan, 2019-01-30:
I have solved this, will be submitting a PR when it's done.

closed 2019-02-03 by trickeydan