srobo/j5 #8: [FEATURE] Hardware Support

trickeydan, 2019-01-30:
We should aim to support hardware.

Specific hardware boards should be listed as separate issues below:

- SR v4 Power Board
- SR v4 Motor Board
- SR v4 Servo Board
- SR Ruggeduino
- SourceBots Arduino / Servo Board

kierdavis, 2019-01-30:
* Webcam + libkoki
* Webcam + apriltag
* Upcoming camera module
* Start zone identification (i.e. competition-mode USB stick)

Presumably RoboCon kit doesn't use the SR power/motor/servo boards, so we should support whatever boards are used instead. I'm not familiar with the kit so I can't list them, unfortunately.

trickeydan, 2019-01-31:
Robocon is best considered as a single board, but is underlyingly more complex and I don't have sufficient knowledge to add it here and be sure about how right I am.

trickeydan, 2019-03-05:
I'm going to close this ticket in order to use the individual tickets.

closed 2019-03-05 by trickeydan