srobo/j5 #23: Use Collections metaclasses to do BoardGroups

trickeydan, 2019-02-03:


kierdavis, 2019-02-04:
Let's wait for a decision on #37 before implementing this, as the outcome will affect which metaclass we use.

kierdavis, 2019-04-19:
The outcome of #37 was that `BoardGroup`s should be ordered (by serial number), so I think [`Sequence`](https://docs.python.org/3/library/collections.abc.html#collections.abc.Sequence) is the right metaclass to use here.

kierdavis, 2019-04-19:
Actually, looking at the current implementation its more like an ordered [`Mapping`](https://docs.python.org/3/library/collections.abc.html#collections.abc.Mapping).

kierdavis, 2019-04-19:
Implementing `collections.abc.Mapping` would require us to change `BoardGroup.__iter__` to iterate over keys (serial numbers) rather than values (boards). Maybe this isn't worth the effort. What do you think @trickeydan?

trickeydan, 2019-04-19:
I think it would be really unintuitive if iteration was over serial numbers rather than boards.

I also agree that collections.abc.Mapping is most appropriate. 

I think this might be a wontfix 😥

kierdavis, 2019-04-19:
We lose out on a few automatically-provided methods, most of which are probably unimportant. The only useful-looking one is `__contains__` which already has its own issue open (#241). I'm happy to call this a wontfix.

Skybound1, 2019-04-20:
I can't think of a class of the top of my head which acts like a dictionary, but iterates over values... so if `__contains__` is the only useful one, I'd agree not fixing and just implementing that is the easiest path to take.

closed 2019-04-21 by trickeydan