srobo/j5 #37: BoardGroup should be modelled on a set, not a list

kierdavis, 2019-02-04:
There is no inherent order to the boards attached to a robot, so we should not be creating an arbitrary order for them by storing them in a list (`BoardGroup.boards`). A set is an unordered collection and would make more sense here.

This is relevant to #23.

If people (I choose you, @trickeydan) think this a good idea then I'm happy to implement this in a PR.

trickeydan, 2019-02-04:
I definitely agree that something should be done here. A set should be used if they are inherently unordered, but I would definitely like to keep the ability to iterate through boards.
I suggest discussing this evening.

kierdavis, 2019-02-04:
It's worth noting that sets can still be iterated over (but the order of iteration should be considered nondeterministic). Let's discuss further this evening as you suggest.

kierdavis, 2019-02-08:
I'm waiting for #25 to be reviewed and merged before starting work on this.

PR #74 (2019-02-15): Sort BoardGroup and fix iteration
Resolves #37 and #38. See commit messages for more details.
commit 9a2413ef (2019-02-15): Order Boards in BoardGroups deterministically

Previously, boards were sorted by the order in which they were
discovered. The discovery order should be considered
nondeterministic, since it depends on things like USB addresses
for hardware backends and is currently undefined for non-hardware
backends.

In an effort to make this more useful to a user, boards will now
be ordered deterministically by sorting the collection by board
serial number.

This order is only visible when iterating through the BoardGroup,
since #25 removed the ability to index boards by integer indices.

Closes #37.
closed 2019-02-15 by kierdavis