srobo/j5 #50: Parameter descriptions and typing

trickeydan, 2019-02-05:
e.g

```python
def format_unit(value, unit):
    """
    Formats the given value as a human readable string using the given units.

    :param float|int value: a numeric value
    :param str unit: the unit for the value (kg, m, etc.)
    :rtype: str
    """
    return '{} {}'.format(value, unit)
```

We should consider this: https://pypi.org/project/sphinx-autodoc-typehints/

sedders123, 2020-07-29:
I'm adding this to the Pre-1.0.0 milestone. My reasoning is that if you call `help()` providing a class, function or property Python will display the relevant docstring. This could help both competitors and implementors to discover more of the API without having to refer to the docs.

Happy to discuss this addition if anyone disagrees.

trickeydan, 2020-08-01:
[Darglint](https://github.com/terrencepreilly/darglint) is likely to be very helpful with this.

I've been unable to configure it exactly how we want, so some upstream contributions may be needed.

PR #725 (2021-08-08): Add param info to docstrings
## Checklist

- [x] I have updated the relevant documentation
- [x] I have added the `semver-major`, `semver-minor` or `semver-patch` label as appropriate
- [ ] I would like multiple reviewers to approve my code before merging

## Why do you want to make these changes?

Better generated API documentation

## Which parts of the codebase do your changes affect?

Most of it!

**Also adds [Darglint](https://github.com/terrencepreilly/darglint) to ensure docstrings are maintained going forward**

## How do your changes affect student or volunteer experience?

Better doc [... cut]
closed 2021-08-08 by trickeydan