srobo/j5 #65: Support SR Ruggeduino

trickeydan, 2019-02-08:
Requires GPIO Pin -  #43 

We need to think about how to support direct communication. Perhaps some sort of serial component?

kierdavis, 2019-02-09:
By direct communication, I assume you mean exposing a channel for competitors' robot code to talk to their custom ruggeduino firmware?

trickeydan, 2019-02-09:
> By direct communication, I assume you mean exposing a channel for competitors' robot code to talk to their custom ruggeduino firmware?

Exactly that.

trickeydan, 2020-07-07:
I think this probably needs breaking up a bit.

- [ ] Generalisation of Arduino-style boards
- [ ] Generalisation of ArduinoBackend
- [ ] Work out how to send custom commands to the Ruggeduino (I think a new Component is in order here)
- [ ] Implementation of RuggeduinoBoard and Backends.

There are three states that a Ruggeduino can be in:

- Standard SR firmware.
- Extended SR firmware (i.e enable command component)
- Completely custom firmware
	- `sr-robot` has an ignore option for this. We should probably add one of these too.
	- Expose a list of ignored serial devices so that students can use them.

PR #609 (2020-07-29): Add Ruggeduino support
## Checklist

- [ ] I have updated the relevant documentation
- [x] I have added the `semver-major`, `semver-minor` or `semver-patch` label as appropriate
- [ ] I would like multiple reviewers to approve my code before merging

## Why do you want to make these changes?
Ruggeduino support is important to have in an API that targets Student Robotics' competition.

## Which parts of the codebase do your changes affect?
`backends/sb/arduino`, `boards/sb/arduino`, `backends/j5/serial` and their tests, plus some new code.

## How do your changes affect student or volunteer exp [... cut]
closed 2020-07-29 by tpoomlmly