srobo/j5 #66: Support SourceBots Servo Assembly

trickeydan, 2019-02-08:
Requires:
- Servos - #42 
- GPIO - #43 
- Ultrasound - This may or may not want to be a component. Consider this when we get here.

kierdavis, 2019-02-13:
I think canonically ultrasound sensors *should* be a component; however this may require a rethink of the component API due to how ultrasound sensors are addressed as a pair of pin numbers (trigger pin and echo pin).

trickeydan, 2019-03-03:
For ultrasound sensors, we could have a 'virtual' `Component`. Two `GPIOPin`s can be passed to it in the constructor?

kierdavis, 2019-03-04:
yeah. There's no reason why we couldn't have high-level components built on top of low-level components. I'm thinking something like:

```python
class ServoAssembly:
  ...
  def ultrasound_sensor(self, trigger_pin, echo_pin):
    if isinstance(trigger_pin, int): trigger_pin = self.gpio[trigger_pin]
    if isinstance(echo_pin, int): echo_pin = self.gpio[echo_pin]
    return UltrasoundProximitySensor(self._backend, trigger_pin, echo_pin)
  ...
class UltrasoundProximitySensor:
  @property
  def distance(self):
    return self._backend.read_ultrasound_sensor(self._trigger_pin.identifier, self._echo_pin.identifier)
```

kierdavis, 2019-08-04:
Ultrasound isn't implemented in the hardware backend yet. Fortunately this is just a matter of extending the existing `SBArduinoHardwareBackend` with `UltrasoundInterface`; pretty much all of the infrastructure is already in place.

kierdavis, 2019-08-05:
#365 fixed this.

closed 2019-08-05 by kierdavis