srobo/j5 #73: [FEATURE] BoardGroup __repr__

kierdavis, 2019-02-13:
`Board` has `__repr__`, so it makes sense for `BoardGroup` to have one too.

trickeydan, 2019-02-14:
#48 is relevant here.

Shivam60, 2019-03-15:
Hey, I would like to take up this issue if no one has been assigned to it.

Shivam60, 2019-03-16:
Are we expecting the following output for the __repr__ ?

BoardGroup(MotorBoard,ServoBoard,Ruggeduin)

trickeydan, 2019-03-16:
The constructor for a `BoardGroup` is:
```python
def __init__(self, board: Type[Board], backend: Backend):
    self.board_class = board
    self._backend: Backend = backend
    self.boards: Dict[str, Board] = OrderedDict()

    self.update_boards()
```
See [here](https://github.com/j5api/j5/blob/3990b4e79add2a5797ccbd089d8ec1311d1aa813/j5/boards/board.py#L81)

#48 gives some info on what a `__repr__` should look like. Probably something like `BoardGroup(PowerBoard, SRv4PowerBoardHardwareBackend`

trickeydan, 2019-03-31:
@Shivam60 How've you been getting on with this? Need any help?

trickeydan, 2019-04-19:
Removing from milestone due to inactive contributor.

@Shivam60 Would you still like to do this?

Shivam60, 2019-04-25:
hey so sorry for my inactivity. I just checked my email's as I was preoccupied with my university exams. I am on it

Shivam60, 2019-04-28:
Guys I am sorry. My laptop's screen broke and I have to travel to different city to get a replacement. It looks like it LL take a week. I am sorry. I request you to remove me from this issue since I won't be able to code till I get a replacement. I am sorry :(

trickeydan, 2019-04-28:
> Guys I am sorry. My laptop's screen broke and I have to travel to different city to get a replacement. It looks like it LL take a week. I am sorry. I request you to remove me from this issue since I won't be able to code till I get a replacement. I am sorry :(

That's alright. Hope everything works out for you and your laptop. Thanks for your help so far :)

PR #303 (2019-06-09): Add __repr__ for BoardGroup.
Fixes #73
commit 1e93d2d2 (2019-06-09): Add __repr__ for BoardGroup.

Fixes #73
closed 2019-06-09 by trickeydan