srobo/j5 #78: Generic component identifiers

kierdavis, 2019-02-13:
Currently different instances of the same type of component on a board (e.g. different power outputs on a power board) are distinguished by an identifier of type `int`. In some situations, it may be more user-friendly to use other types instead.

* *Power outputs*: the SR V4 PB uses labels "H0", "H1", "L0", "L1", "L2" and "L3" to identify its power outputs. Note that other hypothetical boards that make use of the power output component would likely use other labels.
* *Power board LEDs*: the SR V4 PB has two controllable LEDs labelled "run" and "error". It would be nice to be able to use these names (probably in the form of an `Enum`) rather than arbitrarily assigning the identifiers 0 and 1 to these LEDs.
* *Ultrasound sensors*: an ultrasound sensor on the SB servo assembly is addressed by a pair of two pin numbers. However #66 suggests that we're not sure how to model ultrasound sensors in the API yet anyway.

A related facet is that by allowing other types than `int`, we can restrict the possible values of the identifier to solely valid values *at the type level*. For example, where a board only has one instance of a component, we can require the identifier to be of type `NoneType`, which only has one possible value.

Ideally, the type of a component identifier is a function of the both the component type and the board type (to which the component is attached). Unfortunately, my `mypy`-fu is not strong enough to know how to implement this right now.

trickeydan, 2019-02-14:
A possible workaround to this is to define an Enum at the board level. e.g `SRv4PowerBoardPowerOutputLabel`, which by the nature of `Enum` in Python contains a mapping to the integer values whilst also keeping the components nice and generic.

I suggest opening a separate issue for the support of components that use a tuple identifier, e.g ultrasound.

trickeydan, 2019-02-15:
I have put together an example of the `Enum` approach in #93. I agree that it isn't perfect, but it's a start.

trickeydan, 2019-03-03:
I'm going to close this as I think the above solution is good enough. As always, feel free to reopen and discuss.

closed 2019-03-03 by trickeydan