srobo/j5 #79: Specify minimum supported Python version

kierdavis, 2019-02-13:
We should specify what minimum version of Python is required to run j5. Things to consider:

* We likely want to use recent additions to the language, such as:
  * Predictably ordered `dict`s, introduced in 3.7. A concrete use of this feature is highlighted in [this comment](https://github.com/j5api/j5/pull/74#discussion_r256566865).
  * f-strings, introduced in 3.6.
* The Linux distributions on which j5 may be required to run might not provide recent Python versions.
* Read The Docs apparently doesn't support Python 3.7 (#49).

sedders123, 2019-02-13:
Read The Docs [seem to say](https://github.com/rtfd/readthedocs-docker-images/blame/master/README.rst#L30) they support Python 3.7. Was there an error that Read the Docs gave when we tried to use it?

kierdavis, 2019-02-13:
Not sure. #46 also looks relevant.

@trickeydan please can you elaborate?

trickeydan, 2019-02-14:
Read the docs claim they support 3.7, but in reality it doesn't work. That version is only relevant for the building of the docs as far as I can tell.

In terms of distro support, I would be happy to use backports. I think that it is super important to support the latest versions of things whenever possible. I can definitely see a use case for supporting old versions of things if we are aware of a kit that uses it (e.g we don't want to upgrade to 3.8 at the start of the 2020 comp cycle.). I think that we should discuss a versioning mechanism in a separate ticket.

mildlyincompetent, 2019-02-15:
The pip / pipenv configuration currently specifies 3.7, so the repo as a whole is not 3.6-compatible. If we do decide to officially support 3.6 we should ensure that we update the configs to reflect this.

kierdavis, 2019-02-15:
My preference is to stay compatible with 3.6 *for the time being*, to encourage wider adoption of the library. I don't see a problem in increasing the minimum required version when we find we need certain features (or at least, they would make our lives significantly easier), but I don't think its necessary to throw this away *just* yet.

Having said that, our CI runs 3.7 and as @mildlyincompetent points out our Pipfile requires 3.7. I don't even know if the code is compatible with 3.6 at this precise moment. The minimum effort path is thus to stay on 3.7, which I am not opposed to.

trickeydan, 2019-02-15:
To my knowledge, the codebase is entirely compatible with 3.6.
**However**, I would rather support as few versions of python as possible, preferably the most recent stable versions, which happen to be 3.7. When 3.8 is released, we will continue to support 3.7 for some period of time.

trickeydan, 2019-02-15:
I'm happy to support 3.6 if people really want it and believe that there is a case for it, but I would want multiple CI for each version and we need to bear in mind that it only works with CPython 3.6. See #74

trickeydan, 2019-02-17:
Ok, I've been thinking and it's actually a really bad idea to compile our own Python 3.7 for older systems. 

I think supporting 3.6 is a good idea.

trickeydan, 2019-02-17:
Let's make a decision on this tomorrow.

kierdavis, 2019-02-18:
Ah, when you said backports earlier I assumed this was still a distribution-provided package. I didn't realise you meant compiling and distributing it ourself, which I would strongly recommend against as it significantly increases the complexity of deploying the library.

kierdavis, 2019-02-18:
It would be useful to check what versions of Python our consumers actually have available. The SourceBots raspberry pi currently runs raspbian stretch, which contains python 3.5. Despite scouring most of cgit I've not been able to find any indication of what OS version the SR odroid runs, but my gut feeling is that the image likely hasn't been changed since before python 3.6 was released.

trickeydan, 2019-02-18:
> Ah, when you said backports earlier I assumed this was still a distribution-provided package. I didn't realise you meant compiling and distributing it ourself, which I would strongly recommend against as it significantly increases the complexity of deploying the library.

Debian backports has Python 3.7. Alpine does not have backports.

kierdavis, 2019-02-18:
I agree that supporting 3.6 and up is a good idea, let's do this.

closed 2019-02-18 by kierdavis