srobo/j5 #87: [FEATURE] Docker image for circleci

kierdavis, 2019-02-15:
The CircleCI build config performs a few steps that do not depend on the contents of the repository, namely installing `pipenv` and the Code Climate test reporter. It would be nice if these steps didn't have to be re-run every time we push to a branch. Therefore I propose defining a docker image that encapsulates these steps, uploading it to Docker Hub and using it as the base image for the CircleCI build.

The `Dockerfile` would be straightforward, probably looking something like this:

```dockerfile
FROM python:3.7
RUN pip install pipenv
# download test reporter as a static binary
RUN curl -L https://codeclimate.com/downloads/test-reporter/test-reporter-latest-linux-amd64 > ./cc-test-reporter && chmod +x ./cc-test-reporter
```

Bonus points if we use Docker Hub's autobuild, so that the docker image is automatically rebuilt and published upon a push to the repository containing the `Dockerfile` (which, in order to achieve the goal of doing less work upon a push to `j5api/j5`, would probably want to be in a separate repository, say `j5api/docker-ci`).

I am happy to implement this if given the go-ahead.

trickeydan, 2019-02-15:
This sounds like an excellent idea.

kierdavis, 2019-02-15:
@trickeydan I would like to link the Docker Hub account to the @j5-bot github account rather than my personal github account. Please could you either share the credentials for this with me, or let me know your Docker Hub username so I can add you to the org so you can link the accounts yourself?

trickeydan, 2019-02-15:
> @trickeydan I would like to link the Docker Hub account to the @j5-bot github account rather than my personal github account. Please could you either share the credentials for this with me, and/or let me know your Docker Hub username so I can add you to the org so you can link the accounts yourself?

I will message you the creds on Slack

kierdavis, 2019-02-15:
Resolved by #92.

closed 2019-02-15 by kierdavis