srobo/j5 #91: Decide on version control conventions

mildlyincompetent, 2019-02-15:
We need to define a few things regarding how we use version control to ensure contributing is easy and maintains a consistent repo.

Firstly, what branching model are we going for? Currently, we seem to have adopted some version of oneflow / github flow, whereby there is one master branch, and features are developed on short lived feature/hotifx branches with descriptive names. This model appears to work well for us, so I propose that we continue to use it, but that we state as such in the documentation, particularly the documentation new contributors are likely to see. This is especially needed as we now appear to have a bot which automatically deletes merged branches, which is great for this style of development but is completely incompatible with, say, git flow. Therefore we should establish consensus on branching model sooner rather than later.

The second topic is somewhat more controversial: how do we wish to handle merges? Personally I am quite a firm believer in using merge commits with `--no-ff`. The reason for this is it allows for readable history and is very user-friendly; when I look through the git history I can clearly see the branches that commits live on, and I can identify which feature those commits belong to. This does result in merge commits, and some people argue that using rebase merges makes for "nicer history" because it has less merge commits. I disagree; history with less commits is only nicer if it makes for nicer reading. A fast-forward merge obfuscates the fact that there has been a merge and creates confusion for those reading the history - some branches have merge commits, some don't, and it all turns into hot, waxy, pungent garbage. Either way we should reach consensus on how we want to this and communicate this to new contributors.

Lastly, I'd like to discuss "commit quality" on branches. Generally, one should commit little and often (within reason) and then ensure their branched commits are of a reasonable standard (descriptive names, reasonable quantity of changes in each commit, all changes are tightly coupled) before they are merging. My perspective is that rewriting history on branches is acceptable to fix up commits, but that history should be immutable once merged to master, otherwise many nasty things can occur. In general, I don't think we should enforce commit quality beyond very minimal standards (no "fix", "fix?", "FIX" commit crap please), but we should still ensure this is known to new contributors, and potentially provide them with resources to learn git best practices.

Generally I think all of us understand how to and are making a good effort to use version control well, but the only way we can prevent the git history and the state of the repo from turning into a big bag of gore is by ensuring that we are all following the same workflow when we are integrating our contributions (obviously what you do in the privacy of your own branch does not concern me until it gets merged).

I look forward to seeing people's perspectives on this.

trickeydan, 2019-02-15:
> Firstly, what branching model are we going for? Currently, we seem to have adopted some version of oneflow / github flow, whereby there is one master branch, and features are developed on short [...]

Agreed.

> The second topic is somewhat more controversial: how do we wish to handle merges? Personally I am quite a firm believer in using merge commits with `--no-ff`. The reason for this is it allows for [...]

I agree with this statement but remain open to other arguments and do not have strong feelings either way.

> Lastly, I'd like to discuss "commit quality" on branches. Generally, one should commit little and often (within reason) and then ensure their branched commits are of a reasonable standard [...]

Agreed. Force push is blocked on master.

> Generally I think all of us understand how to and are making a good effort to use version control well, but the only way we can prevent the git history and the state of the repo from turning into a [...]

I suggest finding more bots to help with this.

trickeydan, 2019-02-19:
I'm going to block rebase merges in order to implement this.

Feel free to re-open this issue for further discussion as always.

closed 2019-02-19 by trickeydan

kierdavis, 2019-02-19:
Meant to add this earlier but I also agree wholeheartedly with the proposal.