srobo/j5 #97: Should we have competition specific implementations in their own packages?

trickeydan, 2019-02-16:
Should we have competition specific implementations in their own packages? and how far do we go with this?

In my opinion, there should be the following packages for SR as an example:

- `j5` - Core library and component definitions
- `j5.sr` - SR Board and backend definitions.
- `sr.robot` - The actual API that competitors interact with.

kierdavis, 2019-02-18:
I agree fully with your proposed structure. However I don't think it's worth splitting it up while we're still in such an early stage of development since maintaining code, CI, issues and pull requests across multiple repositories adds a layer of complexity that I feel is unnecessary right now. My suggestion would be to hang on until most things are implemented and we're confident that the project structure isn't going to radically change.

trickeydan, 2019-02-18:
Ok, I'm marking this as blocked on us writing a lot more code. I realise that this is a somewhat arbitrary thing to block on, please feel free to comment on this issue as necessary.

kierdavis, 2019-02-19:
I have updated the title to reflect the new direction the issue has taken.

mildlyincompetent, 2019-02-24:
M o n o r e p o

There is no _technical_ reason that we couldn't have separate python packages in one repo.

That said, I agree with @kierdavis until such time as separate packages become genuinely needed.

trickeydan, 2019-08-26:
Relevant: #149

trickeydan, 2019-08-26:
> M o n o r e p o

I'm unsure if you're joking, but a monorepo is probably not a bad idea.

There should still be separate python packages though

kierdavis, 2019-08-26:
Assuming that we're not intending to transfer maintainership of SR boards to SR's developers (and likewise for other competitions and their specific hardware), I'm onboard for monorepo.

Also note that we should avoid considering boards to be competition-specific. Anyone is free to set up a new competition using whichever boards they choose, and doing so shouldn't incite us to change our package structure. It makes sense to group boards into packages by the principal designer of the board (whether that be hardware/firmware/software), though.

trickeydan, 2022-04-29:
Following the successful deployment of `j5_zoloto`, I believe consensus has been reached here.

closed 2022-04-29 by trickeydan