srobo/j5 #108: Logging Support

trickeydan, 2019-02-19:
We need support for logging. This should be done with the standard python `logging` library.

For now let's put it on `stdout`. Putting it in `journalctl` and on the usb drives can be worried about when we implement USB drives.

trickeydan, 2019-03-01:
This could be useful for the test scripts, e.g just set the `LOGGING_LEVEL` to `DEBUG`

trickeydan, 2019-06-09:
@sedders123 has made a start on this I believe.

kierdavis, 2019-08-06:
I'm not sure what the scope of this issue is, but `sbot` already has some pretty good logging of software and firmware versions, and #369 will add logging of operations performed on components. I can't think of any other major things that should be logged.

trickeydan, 2019-11-26:
This issue was technically completely, and I would argue is superseded by #370

closed 2019-11-26 by trickeydan