srobo/j5 #115: Do we need the 'demo' board?

trickeydan, 2019-02-19:
I initially created the demo board to help me build the structure of the project. After #93 is merged, we will have a full working example of a board, thus potentially making the `DemoBoard` redundant.

Is it worth removing this in favour of documentation of the `Board` definition and implementation process?

kierdavis, 2019-02-21:
I agree that a real, well-defined `Board` serves as better documentation than a hypothetical `Board` with a blurry specification. Additionally, the fact that the power board is used in actual kits means its much harder for the code to become outdated without us knowing about it.

trickeydan, 2019-02-21:
Blocked on #93

PR #180 (2019-03-12): Remove Demo Board and Backend
Fixes #115 and #158
closed 2019-03-12 by trickeydan