srobo/j5 #116: Redundant reference to Backend is passed during Component construction

trickeydan, 2019-02-19:
e.g
```python
    def __init__(self, identifier: int, board: Board, backend: ServoInterface) -> None:
        self._board = board
        self._backend = backend
        self._identifier = identifier
```

We  then call `self._backend.get_servo_position(self._board, self._identifier)`, but we could access this on `board`.

We would have to fiddle with the types to remove this though.

kierdavis, 2019-02-21:
Perhaps its the board that's redundant, not the backend? As far as I can recall, boards don't expose methods like `get_servo_position` - they just expose a collection of servo components which talk directly to the implementor of the component's backend interface.

trickeydan, 2019-02-21:
We should wait until #118 is fixed before continuing on this issue.

trickeydan, 2019-03-03:
We've discussed this before and I'm going to list the changes that we should make for this:

- `Backend` is now specific to a board instance. Every board instance has it's own Backend instance.
- Thus we don't need to pass `Board` to calls on `Backend` instances as a reference is stored on the `Backend`

Incidentally, this will also fix #153.

kierdavis, 2019-03-12:
I agree, and I think this is ready to be implemented.

This will also fix #186.

trickeydan, 2019-03-31:
Fixed by #195

closed 2019-03-31 by trickeydan