srobo/j5 #118: Backend-to-Board correspondence

kierdavis, 2019-02-20:
Are we expecting there to be one instance of `Backend` corresponding to each instance of `Board`, or for there to be a singleton instance of `Backend` that operates on any `Board`? To put it another way, should `Backend` objects store board-specific information (e.g. USB device handle) in their attributes, or should this information be stored on the `Board`?

If the former, we need to review how we create and use `Backend`s. My understanding is that our flow currently looks like this (please correct me if I'm wrong):
* `Robot.__init__` calls `Environment.get_backend(board_class)`, obtaining `backend`, a new instance of `backend_class`
* `Robot.__init__` creates a `BoardGroup`, passing `backend`
* `BoardGroup.__init__` calls `BoardGroup.update_boards`, which calls `backend.discover()`
* `backend.discover` creates and returns a number of `board_class` instances, passing the serial number and `self.environment` to the constructor. It is undefined where backend-specific data about the board is kept
* `Board.__init__` calls `Environment.get_backend(self.__class__)`, creating another instance of `backend_class` on which component operations for that board will be performed
* `BoardGroup.update_boards` stores the collection of `Board` instances

I would expect the flow to look something more like:
* `Robot.__init__` calls `Environment.get_backend(board_class)`, obtaining `backend`, **the only instance** of `backend_class` (using the [singleton pattern](https://en.wikipedia.org/wiki/Singleton_pattern))
* `Robot.__init__` creates a `BoardGroup`, passing the `backend`
* `BoardGroup.__init__` calls `BoardGroup.update_boards`, which calls `backend.discover()`
* `backend.discover` creates and returns a number of `board_class` instances, passing the serial number, **the backend** to the constructor. Backend-specific data about the board is held by the backend instance, in a dict keyed by board instance.
* `BoardGroup.update_boards` stores the collection of `board_class` instances

However, if we want to have one `Backend` per `Board`, the flow should probably look something like this:
* `Robot.__init__` calls `Environment.get_backend(board_class)`, obtaining **the corresponding `Backend` class**
* `Robot.__init__` creates a `BoardGroup`, passing the **backend class**
* `BoardGroup.__init__` calls `BoardGroup.update_boards`, which calls `backend_class.discover()`
* `backend_class.discover` creates a number of `backend_class` instances, which can hold arbitrary backend-specific data about the corresponding board
* `backend_class.discover` wraps each `backend_class` instance with a `board_class` instance.
* `BoardGroup.update_boards` stores the collection of `board_class` instances

What are your thoughts on these approaches?

trickeydan, 2019-02-20:
I originally designed it with one `Backend` instance per `Board` instance in mind such that specific information such as usb device handle can be stored on the `Backend`.

I personally think having a Singleton with a `dict` of board info is quite messy, so would prefer the other approach. I am open to it if it makes sense though.

kierdavis, 2019-02-21:
One `Backend` per `Board` makes most sense to me too. In that case, `Environment.get_backend` should probably return a class rather than an instance.

PR #123 (2019-02-21): Make Environment.get_backend return a class
Resolves #118.
commit cba7d34d (2019-02-21): Make Environment.get_backend return a class

Resolves #118.
closed 2019-02-21 by kierdavis