srobo/j5 #120: Cross-Platform Compatibility

trickeydan, 2019-02-20:
I'm currently working on #3 and I have a choice of implementations, one of which I would prefer but would also prevent the library working on Windows. I think it is really important to try and keep this cross platform if possible. Currently there is nothing Unix-specific.

I would be interested in the views of others.

sedders123, 2019-02-20:
I think that we should strive for cross-platform compatibility until we reach a problem that has no other solution but to support just the one platform (or the multiplatform solution requires *substantially* more work).

According to this [site](https://netmarketshare.com/operating-system-market-share.aspx?options=%7B%22filter%22%3A%7B%22%24and%22%3A%5B%7B%22deviceType%22%3A%7B%22%24in%22%3A%5B%22Desktop%2Flaptop%22%5D%7D%7D%5D%7D%2C%22dateLabel%22%3A%22Trend%22%2C%22attributes%22%3A%22share%22%2C%22group%22%3A%22platform%22%2C%22sort%22%3A%7B%22share%22%3A-1%7D%2C%22id%22%3A%22platformsDesktop%22%2C%22dateInterval%22%3A%22Monthly%22%2C%22dateStart%22%3A%222018-02%22%2C%22dateEnd%22%3A%222019-01%22%2C%22segments%22%3A%22-1000%22%7D) Windows has an 87% market share, and while I'm sure in tech circles that number changes it seems crazy to me that we should alienate such a large contributor pool when we don't have to.

I fully understand there are some things that can't be done cross platform, but when we get to a case of this I reckon we should clearly document it, try to make sure that it doesn't make the *whole* project unusable, and try to point out suitable workarounds if there are any (compat libraries, instructions for Windows Sub System for Linux etc.)

kierdavis, 2019-02-21:
Also note that a j5-based simulator must be able to run on Windows.

My preference would be for a solution that works on all platforms, but choosing between a Unix-specific solution and a Windows-specific solution at runtime would be acceptable.

trickeydan, 2019-02-22:
I think this issue has been resolved at this point and I shall be implementing a cross-platform solution.

closed 2019-02-22 by trickeydan