srobo/j5 #125: Piezo frequencies could be floats

kierdavis, 2019-02-21:
Hypothetically, piezo implementations may support more precision than an integer allows, and the frequencies of musical notes we provide in the `Note` enum could be specified to more precision than they currently are. It seems reasonable to use floats everywhere and make it the responsibility of the backend to round to an integer if it does not support fractional frequencies.

This should probably be implemented before or at the same time that #122 is, to avoid having to change things twice.

trickeydan, 2019-02-21:
In order to not break `Note` being of type `IntEnum` when it is moved to the `Piezo` component (#122), we would need to support `Union[int,float]`

kierdavis, 2019-02-21:
Ideally `Note` shouldn't have to be an `IntEnum`, because frequencies should be real numbers rather than integers. It sucks that we have to use this workaround to make the typechecking pass (see [this thread](https://github.com/j5api/j5/pull/119#discussion_r258686925) for reference).

trickeydan, 2019-03-17:
@HU90m I feel like you might want this one too?

PR #243 (2019-04-19): Use floating point numbers for Piezo
Resolves #125
closed 2019-04-19 by trickeydan