srobo/j5 #146: Add setter for Component that throws error

trickeydan, 2019-03-01:
This will stop users accidentally overriding components.

e.g

```

r.power_board.led = True
```

trickeydan, 2019-03-03:
This isn't as easy as I thought. [See here](https://stackoverflow.com/questions/11024646/is-it-possible-to-overload-python-assignment)

kierdavis, 2019-03-03:
Things like `r.power_board.start_button` are properties, right?

```python
@property
def start_button(self) -> Button:
  return self._start_button
@start_button.setter
def start_button(self, value: Any):
  raise Exception("no")
```

trickeydan, 2019-03-03:
> Things like `r.power_board.start_button` are properties, right?

Yeah. Would be nice if we can we `__set__` so that we don't have to define a `@property.setter` for every component.

kierdavis, 2019-03-04:
Looks like this is already taken care of - properties appear to be read-only if you don't define a setter:

```
>>> class Foo(object):
...   def __init__(self):
...     self._x = 0
...   @property
...   def x(self):
...     return self._x
... 
>>> foo = Foo()
>>> foo.x
0
>>> foo.x = 2
Traceback (most recent call last):
  File "<stdin>", line 1, in <module>
AttributeError: can't set attribute
```

Indeed, this holds true for our own code:

```
>>> from j5.boards.sr.v4 import PowerBoard
>>> board = PowerBoard(None, None)
>>> board.start_button
<j5.components.button.Button object at 0x7fdeb61ecf60>
>>> board.start_button = True
Traceback (most recent call last):
  File "<stdin>", line 1, in <module>
AttributeError: can't set attribute
```

closed 2019-03-04 by Skybound1