srobo/j5 #148: Do we want to keep the CLA?

trickeydan, 2019-03-02:
Several people today (@RealOrangeOne, @mildlyincompetent) have raised issues today about the CLA on this repo and think that it's unnecessary.

trickeydan, 2019-03-05:
@mildlyincompetent Please could you make a case for removing the CLA?

__PROS of the CLA__:

- Ensures that the licence agreement has been explicitly read and agreed to.
- A bot ensures that this is the case
- This will ensure that contributors do not try to retract their work or lead to licensing disputes in the future.

Options if we decide not to keep the CLA:
- Replace the CLA text with the LICENCE to ensure that contributors have read it.
- Remove both the CLA and the Bot.

[CLA Text](https://gist.github.com/trickeydan/e1a7b8d726846adf50ca84a2bcccedc2)

mildlyincompetent, 2019-03-05:
First let's discuss the pros of the CLA to see how advantageous they are:

> Ensures that the licence agreement has been explicitly read and agreed to.

Realistically, anyone contributing to an open source project will understand that their contributions will be shared under the license of the project. It is very difficult to contribute to a GitHub project without noticing the big flashy license. Therefore we can pretty much assume that, for good faith contributors, the CLA serves no informational or consent-establishing purpose.

> A bot ensures that this is the case

This is better than requiring a human to ensure it, but still adds friction to contributions.

> This will ensure that contributors do not try to retract their work

Realistically this does not protect us adequately from malicious (intentionally or otherwise) contributors. The MIT license (which is commonly accepted as applying to contributions) already grants us a license to use contributions. But let's consider the two primary risks: malicious contributor, and malicious employer.

Malicious contributor: they hit us with a big smelly lawsuit. If they DMCA us (which they could since GitHub is based in the US) or just throw lots of money at the problem, we're basically screwed regardless unless we have some hidden pot of money for lawyers that I am unaware of. Should we fight a legal case on this topic, the CLA simply makes a win more likely - it does not ensure it, and nor does its lack ensure a loss. I would sure hope that a reasonable court would side with us regardless, however given that we'd be dealing with potentially international copyright law, I very strongly doubt we could actually afford such a case.

Malicious employer: the CLA does effectively nothing to protect us, since the contributor's IP is owned by their employer not the contributor. They do not have the authority to either sign the CLA or release their work under the MIT license. We are pretty much screwed in this instance. Yes the CLA requires them to confirm that they have this authority, but that doesn't really help us if this happens to be untrue.

> or lead to licensing disputes in the future.

Personally I consider this a disadvantage. Being able to sneakily re-license the whole thing under the SSPL feels like a bad thing. The need to avoid licensing disputes is easily resolved by just picking a good license and sticking to it, but that's just my opinion.

Note that if the copyright is assigned to "j5 contributors", it is unclear whether we can actually trivially re-license anyway, so if this is a significant concern, I'm not sure if our current approach to the CLA is appropriate.

I'd like to add that our CLA in general is a bit murky - we state that "to the extent possible under law, you transfer all copyright and related or neighboring rights of the code or documents you contribute to the project itself or its maintainers." The project itself is not, to my knowledge, a legal entity. Therefore we are left with the project maintainers, who are not really well defined should this come to a legal dispute. I really don't think it gives us that much protection.

So what are the disadvantages of a CLA:

Easy re-licensing: if the CLA has an appropriate copyright transfer clause (I'm not entirely convinced ours does), the project can be easily re-licensed. I consider this a disadvantage precisely because the maintainers alone can re-license, including under nasty licenses. Since we're using MIT, there's little in the way of simply forking and licensing under any more restrictive license, but still - I don't really like the idea of contributors effectively having no rights as to how their work is licensed.

Contribution friction: this is a direct and active disadvantage. In general I feel that this project is currently quite hostile to new contributors, and this doesn't help. I myself am not convinced that any of the advantages of the CLA are either that advantageous or easily achieved with our current CLA. Meanwhile the friction generated for contributors exists as soon as someone wants to contribute. A lot of people are scared of big legal agreements, or simply get confused.

Personally I would support removing the CLA and the bot. I do not support the replacement of the CLA with the LICENSE file, because that keeps the main real disadvantage of friction, while just weakening basically all of the perceived advantages.

trickeydan, 2019-03-06:
> Realistically, anyone contributing to an open source project will understand that their contributions will be shared under the license of the project. It is very difficult to contribute to a GitHub [...]

I dislike assumptions.

> This is better than requiring a human to ensure it, but still adds friction to contributions.

> Personally I consider this a disadvantage. Being able to sneakily re-license the whole thing under the SSPL feels like a bad thing. The need to avoid licensing disputes is easily resolved by just picking a good license and sticking to it, but that's just my opinion.

The lack of a CLA doesn't stop this either. It's still under MIT and you can still fork.

> Note that if the copyright is assigned to "j5 contributors", it is unclear whether we can actually trivially re-license anyway, so if this is a significant concern, I'm not sure if our current approach to the CLA is appropriate.

We should probably define this better. I suggest looking at other MIT licence projects and doing what they do.

> Easy re-licensing: if the CLA has an appropriate copyright transfer clause (I'm not entirely convinced ours does), the project can be easily re-licensed. I consider this a disadvantage precisely [...]

Forking can be done anyway.

>  In general I feel that this project is currently quite hostile to new contributors

If this is the case, then I am greatly concerned and wish to discuss the issues that you see that cause this along with any proposed solutions.

> Personally I would support removing the CLA and the bot.

🤖😢😢🤖

To be clear. I agree with the removal of it at this stage.

kierdavis, 2019-03-06:
I do think its important that our contributors are made aware that their contributions are being made on the terms of the license, because many of our contributors are less familiar with open source development practices and software licensing than you or I and it would be good to help them learn more about it. However, I don't feel too strongly about how we go about doing this. I'm happy if you want to remove the CLA. Personally I'd happy with the statement "by contributing to this project I agree that my contributions will be made public under the terms of the project's license" being made visible when creating a PR, or simply in the contributing guidelines.

RealOrangeOne, 2019-03-06:
I only signed the CLA because I believed that my contribution was more important than my objections to a CLA. Personally I consider the requirement to sign a CLA a blocker to me contributing. A project on this scale doesn't need a CLA, most of the contributors are doing so on good faith.

Entities like Facebook require CLAs to protect them against any legal issues, which would be a massive deal for them.

The main thing a CLA protects against in our case is re-licensing (I think). The chances of us doing that are incredibly low, and even then there will be a large intersection between the people involved in said discussion and the contributors.

trickeydan, 2019-03-07:
I have disabled the CLA. Please re-open the ticket if we want it reinstated.

closed 2019-03-07 by trickeydan