srobo/j5 #149: Replace metaclass with class decorators

trickeydan, 2019-03-02:
We need to keep the auto-registration so that external modules can define `Board`s and have them register with the appropriate `Environment`.

@prophile is probably a good person to poke with this.

trickeydan, 2019-03-12:
The solution to this issue should also fix the issue where the backend has to be imported, but not directly used, resulting in messy code and a linting error

trickeydan, 2019-03-31:
I'm delaying this to v0.0.4

trickeydan, 2019-06-09:
This still needs doing, but there are other things that are blocking shipping for Smallpiece. We can deal with slightly messy code for a bit.

trickeydan, 2019-08-26:
I think what the fundamental issue here is to do with auto-registration of classes. That's what the issue is.

I've done some research into plugin frameworks and none of them do exactly what we want. 

I think that explicit registration of modules is the correct approach to take.

kierdavis, 2019-08-26:
> I think that explicit registration of modules is the correct approach to take.

Agreed. It's also less magical and thus more Pythonic.

closed 2019-09-10 by trickeydan