srobo/j5 #165: Environment Metadata Support

trickeydan, 2019-03-05:
- Start Zone
- `DEV` / `COMP` mode.

This has been implemented in the past using USB drives. I suggest we do this same, but have the ability to support multiple backends and arbitrary attributes.

This is probably best implemented as a `Component`, and a 'Virtual' `Board`.

trickeydan, 2019-03-12:
Moving this back to v0.3 because v0.2 is a clean up release

trickeydan, 2019-06-06:
This should probably leave it down to API Authors to specify a source of the data.

`sbot` will probably want to load using a glob in the directories that are managed by `usbmount`.

kierdavis, 2019-06-14:
@trickeydan It's just occurred to me that maybe this doesn't need to be a responsibility of `j5` at all, since it's very competition-specific. Maybe it should just be implemented in `sbot` etc. instead?

trickeydan, 2019-06-15:
> @trickeydan It's just occurred to me that maybe this doesn't need to be a responsibility of `j5` at all, since it's very competition-specific. Maybe it should just be implemented in `sbot` etc. instead?

This is a fair point actually. When this issue was created, `sbot` wasn't a thing yet so I don't think we thought about that.

trickeydan, 2019-06-15:
Another thought on this: Currently we have a `Dict[str,str]` type for the metadata. Leaving this down to the API author removes this restriction...

I think I might be in favour of this being _out of scope_ for `j5`.

kierdavis, 2019-06-15:
Let's move this out of j5 for now. Once its implemented in at least two places (e.g. `sbot` and the SR equivalent), we'll reconsider this to see if there's any common code that's worth factoring out into j5.

closed 2019-06-16 by trickeydan