srobo/j5 #175: nix-shell does not work

mildlyincompetent, 2019-03-11:
I run command

`nix-shell --arg pkgsSrc 'fetchTarball https://nixos.org/channels/nixpkgs-unstable/nixexprs.tar.xz'`

It do:

```
unpacking 'https://nixos.org/channels/nixpkgs-unstable/nixexprs.tar.xz'...
these derivations will be built:
  /nix/store/lw70s6f11n56bbakpkdwzqg5y7wdjpiq-python3.7-poetry-0.12.10.drv
these paths will be fetched (69.88 MiB download, 309.76 MiB unpacked):
  /nix/store/04qyi1m085da5qkq1bs1fmyybr08095c-python3.7-pylev-1.3.0
  /nix/store/0j1sc30kjf9b3j7j0sp68jns2v34apr0-pcre-8.42
  /nix/store/0y7jmqnj48ikjh37n3dl9kqw9hnn68nq-binutils-2.31.1
  /nix/store/1cgqx3lzd749lr43cx31sy1pdm606zh3-python3.7-tomlkit-0.5.3
  /nix/store/2h8df1ppf9ypgggzigavsf02idhddzdn-systemd-239.20190219-lib
  /nix/store/2mfzxcsf6cbfwgklai0jb3lfc4m02i0c-python3.7-certifi-2018.11.29
  /nix/store/2nvbrjns957jnggy8afn1skb3a1j1hwk-libgcrypt-1.8.4
  /nix/store/2rnkpfpyd88nhnccr3s08rb9w8dpxy92-python3.7-jsonschema-3.0.0a3
  /nix/store/2xb0wphs5h89i0k1r8lya9fpsjwkn7xm-openssl-1.0.2r
  /nix/store/2zs46xf1cf621bc6czrdhzc1yf7dffm8-python3.7-pycparser-2.19
  /nix/store/386iyhs7rj5qlv981lfv72sqz0m083xc-python3.7-shellingham-1.2.8
  /nix/store/3hrcvnbsar9qalacvp9xr08bkz9hfxs4-python3.7-pyparsing-2.3.1
  /nix/store/47lrl9p27k0a385kyk6hzad7h5nkzlw1-python3.7-pyOpenSSL-19.0.0-dev
  /nix/store/5an6j4hybn8pvamva8mdc577v1237y01-python3.7-pyrsistent-0.14.11
  /nix/store/5lyvydxv0w4f2s1ba84pjlbpvqkgn1ni-linux-headers-4.19.16
  /nix/store/63karsgdg7fm3q0if4zfd7apbd8ac1ci-stdenv-linux
  /nix/store/681354n3k44r8z90m35hm8945vsp95h1-glibc-2.27
  /nix/store/6ny351q713xmkn4svbl8fy9klhjaw948-python3.7-msgpack-python-0.6.1
  /nix/store/7xmnp58nskq5im960xdbw5x7461wssm3-python3.7-six-1.12.0
  /nix/store/8a36yi82pxfykpli6cra8w8p93nl7r4s-python3.7-cffi-1.12.1-dev
  /nix/store/8n7v99ii7cg694pz0cfch422zvkdqfv4-gnumake-4.2.1
  /nix/store/92pgv1sqdrapglnqzjqgjhpf91p6r7qr-python3.7-packaging-19.0
  /nix/store/9a1rx8j84zl0aygwbk5m1mvxzfgswxym-libcap-2.26-lib
  /nix/store/a0s0xn0z52jmnlhlijs1q628h8y262ll-python3.7-requests-toolbelt-0.9.1
  /nix/store/a590ac76ynxc0hhpwgvwg1fzlq2j57m5-hook
  /nix/store/b11xm5gvhrdnx76fi7ph1k07a133qz29-python3.7-cryptography-2.5
  /nix/store/bl1cy3wni09zif764s948mdpsaqp0xqf-hook
  /nix/store/bl8lqpixn7nwywaynbv6cbhszp875x23-python3.7-memcached-1.51-1.51
  /nix/store/c7z3y86fb7w3j4lz9b3xd4q70j17m8wd-python3.7-pastel-0.1.0
  /nix/store/ci779zkwy8dz9snbd2zgjdrqkh50s4nq-python3.7-pkginfo-1.5.0.1
  /nix/store/cinw572b38aln37glr0zb8lxwrgaffl4-bash-4.4-p23
  /nix/store/cwhwjx1nrdnv61cnhfwm70j9sq1dcncy-python3.7-pysocks-1.6.8
  /nix/store/d4n93jn9fdq8fkmkm1q8f32lfagvibjk-gcc-7.4.0
  /nix/store/d9s1kq1bnwqgxwcvv4zrc36ysnxg8gv7-coreutils-8.30
  /nix/store/di576afdwar4h5yriffmr3pskys27wkf-python3.7-urllib3-1.24.1
  /nix/store/f5wl80zkrd3fc1jxsljmnpn7y02lz6v1-glibc-2.27-bin
  /nix/store/g4rgyrkq7yfnqfmv5hxsc32x6cydh1nc-sqlite-3.26.0
  /nix/store/g8pkjkl01m87bxklp35iwg0wnliby0bz-poetry-0.12.10.tar.gz
  /nix/store/gbwx5sc4w0i0ipnmsa0nsk88svs983vl-python3-3.7.2
  /nix/store/ggd6cq5hybxblmhyzbziihd6zl8b8ccj-python3.7-requests-2.21.0
  /nix/store/ghzg4kg0sjif58smj2lfm2bdvjwim85y-gcc-wrapper-7.4.0
  /nix/store/gyn43s41wdcsbcb75jgxg49cwmlqb5jx-libusb-1.0.22
  /nix/store/hfhg862bl9jgx946ywcg361jl2l1gp1b-python3.7-html5lib-1.0.1
  /nix/store/hl78qs2ziasikqq6lwhnn2misq1rngzk-python3.7-asn1crypto-0.24.0
  /nix/store/hlnxw4k6931bachvg5sv0cyaissimswb-gcc-7.4.0-lib
  /nix/store/hqjvxjk61nsv2rlpb2d5ry3miipy9a9h-python3.7-setuptools-40.8.0
  /nix/store/hql9ki8x230src24ljb9fad7rxxpzal0-attr-2.4.48
  /nix/store/ial8a56ys2rswfq4j2abm3cl99li5b6p-bash-interactive-4.4-p23-dev
  /nix/store/iiymx8j7nlar3gc23lfkcscvr61fng8s-zlib-1.2.11
  /nix/store/ikk8jjnhhgvg0asid5m0k7jwyf8qkl7l-libffi-3.2.1-dev
  /nix/store/ix2q5vnynzfgsi4nqj8pmcbx2kf78wrm-gzip-1.10
  /nix/store/j3i47v2f9jvdink4c2xmrplg8zbkkc98-python3.7-clikit-0.2.3
  /nix/store/jl17v3f872ax5wfi49b04mjzjz10rnra-patch-2.7.6
  /nix/store/jmzfljmwzv5q29zhrxb7ikf7m0zggv4i-hook
  /nix/store/k0y9nrym785l61pjw0lp0qk90v2nbpqs-ncurses-6.1-20190112
  /nix/store/kbbiky87bsdndh63m7919sxkf3gsh5zq-libffi-3.2.1
  /nix/store/kmxlgixizhxv8llspci6l1j9dh2gpnam-gdbm-1.18.1
  /nix/store/kpxyydafq0b1yfyqachvzd9vg3bsk9si-python3.7-msgpack-0.6.1
  /nix/store/krhqmaqal0gklh15rs2bwrqzz8mg9lrn-findutils-4.6.0
[... 686 lines cut ...]
adding 'poetry-0.12.10.dist-info/entry_points.txt'
adding 'poetry-0.12.10.dist-info/top_level.txt'
adding 'poetry-0.12.10.dist-info/RECORD'
removing build/bdist.linux-x86_64/wheel
installing
/build/poetry-0.12.10/dist /build/poetry-0.12.10
Processing ./poetry-0.12.10-py3-none-any.whl
Requirement already satisfied: cachecontrol[filecache]<0.13.0,>=0.12.4 in /nix/store/paz52xm8whyal9vklzys9ak2d3mv8ziv-python3.7-CacheControl-0.12.5/lib/python3.7/site-packages (from poetry==0.12.10) (0.12.5)
Requirement already satisfied: tomlkit<0.6.0,>=0.5.1 in /nix/store/1cgqx3lzd749lr43cx31sy1pdm606zh3-python3.7-tomlkit-0.5.3/lib/python3.7/site-packages (from poetry==0.12.10) (0.5.3)
Requirement already satisfied: cleo<0.7.0,>=0.6.7 in /nix/store/p77fayqkfkkfllgly0dj79lkfqynwrgv-python3.7-cleo-0.7.2/lib/python3.7/site-packages (from poetry==0.12.10) (0.6.8)
Requirement already satisfied: cachy<0.3,>=0.2 in /nix/store/yr0jnxwrlx8pcndxynh5g6hqjkiaiqi5-python3.7-cachy-0.2.0/lib/python3.7/site-packages (from poetry==0.12.10) (0.2.0)
Requirement already satisfied: html5lib<2.0,>=1.0 in /nix/store/hfhg862bl9jgx946ywcg361jl2l1gp1b-python3.7-html5lib-1.0.1/lib/python3.7/site-packages (from poetry==0.12.10) (1.0.1)
Requirement already satisfied: shellingham<2.0,>=1.1 in /nix/store/386iyhs7rj5qlv981lfv72sqz0m083xc-python3.7-shellingham-1.2.8/lib/python3.7/site-packages (from poetry==0.12.10) (1.2.8)
Requirement already satisfied: pyparsing<3.0,>=2.2 in /nix/store/3hrcvnbsar9qalacvp9xr08bkz9hfxs4-python3.7-pyparsing-2.3.1/lib/python3.7/site-packages (from poetry==0.12.10) (2.3.1)
Requirement already satisfied: jsonschema<4.0,>=3.0.0a3 in /nix/store/2rnkpfpyd88nhnccr3s08rb9w8dpxy92-python3.7-jsonschema-3.0.0a3/lib/python3.7/site-packages (from poetry==0.12.10) (3.0.0a3)
Requirement already satisfied: pyrsistent<0.15.0,>=0.14.2 in /nix/store/5an6j4hybn8pvamva8mdc577v1237y01-python3.7-pyrsistent-0.14.11/lib/python3.7/site-packages (from poetry==0.12.10) (0.14.11)
Requirement already satisfied: pkginfo<2.0,>=1.4 in /nix/store/ci779zkwy8dz9snbd2zgjdrqkh50s4nq-python3.7-pkginfo-1.5.0.1/lib/python3.7/site-packages (from poetry==0.12.10) (1.5.0.1)
Collecting requests-toolbelt<0.9.0,>=0.8.0 (from poetry==0.12.10)
  Could not find a version that satisfies the requirement requests-toolbelt<0.9.0,>=0.8.0 (from poetry==0.12.10) (from versions: )
No matching distribution found for requests-toolbelt<0.9.0,>=0.8.0 (from poetry==0.12.10)
builder for '/nix/store/lw70s6f11n56bbakpkdwzqg5y7wdjpiq-python3.7-poetry-0.12.10.drv' failed with exit code 1
error: build of '/nix/store/lw70s6f11n56bbakpkdwzqg5y7wdjpiq-python3.7-poetry-0.12.10.drv' failed
```

Oman I am not good with help pls to computer

kierdavis, 2019-03-11:
This is reproducible at the moment. The "nixpkgs-unstable" channel lags slightly behind the github master branch, likely because Hydra is busy evaluating things. Running

```
nix-shell --arg pkgsSrc 'fetchTarball https://github.com/NixOS/nixpkgs/archive/master.tar.gz'
```

should give you the behaviour you desire. I'll update the `shell.nix` to use this URL instead.

kierdavis, 2019-03-11:
#176.

commit 19df2e2a (2019-03-11): shell.nix: use a more up-to-date URL in workaround command

Fixes #175.
closed 2019-03-11 by trickeydan