srobo/j5 #196: Fix 'Using type() instead of isinstance() for a typecheck.' issue in tests\backends\hardware\test_env.py

trickeydan, 2019-03-12:
[CodeFactor](https://www.codefactor.io/repository/github/j5api/j5) found an issue: Using type() instead of isinstance() for a typecheck.

It's currently on:
[tests\backends\hardware\test_env.py:9
](https://www.codefactor.io/repository/github/j5api/j5/source/master/tests/backends/hardware/test_env.py#L9)Commit 85a5ddb8c6693fbf43b85891bf36ced1de3320f9

Shivam60, 2019-03-15:
Greetings, Thanks for marking it good first issue. I would like to contribute to resolving this issue.

trickeydan, 2019-03-15:
Hi @Shivam60,
Thanks for your help! We welcome all people to our projects, no matter their experience or ability.

I can see that you've already forked the repository. If you make your changes in their and then create a pull request against this repository `master`, we can then review the code, run automated tests, approve and merge.

commit 3990b4e7 (2019-03-16): Merge pull request #200 from Shivam60/pr/196

Fixes #196
closed 2019-03-16 by Skybound1