srobo/j5 #249: BoardGroup.singular uses Exception

trickeydan, 2019-04-18:
File "/home/dan/.cache/pypoetry/virtualenvs/sbot-py3.7/lib/python3.7/site-packages/j5/boards/board.py", line 94, in singular
    raise Exception("There is more than one or zero boards connected.")

This should be a more specific error.

kierdavis, 2019-04-18:
I don't think any of the standard python Exceptions describe this scenario very well, so I think it's sensible to define our own exception type for this. I don't have a good name for it though.

kierdavis, 2019-04-18:
Somewhat related: I find this error message quite grammatically unclear. Perhaps something like

> Expected exactly one board to be connected

would be clearer? Or even:

> Expected exactly one `{type(board).__name__}` board to be connected

trickeydan, 2019-04-19:
Given that we have the `board.name` attribute, I agree.

trickeydan, 2019-09-16:
This was fixed in #336

closed 2019-09-16 by trickeydan