srobo/j5 #250: Should "safe state" of motor be BRAKE or COAST?

kierdavis, 2019-04-18:
Spun out of #227; see [discussion](https://github.com/j5api/j5/pull/227#discussion_r275446457) there.

A third option is to leave this decision down to the competition vendor to decide.

mildlyincompetent, 2019-04-25:
Generally speaking, when someone is stabbed, it is best practice *not* to remove the stabbing instrument prior to medical personnel being present, the reason for this being simply that the removal adds further movement to a stab wound, which generally causes bad things to occur.

Similarly, if one is working with a giant, deadly, spinny sawblade of death, you would much rather it come to an abrupt stop if something goes wrong as opposed to coasting its way through your arm and/or face.

Now of course, the robots we use do not generally carry stabbing paraphernalia or spinny sawblades of death, but the general point I feel stands - if your hand is trapped in a mechanism, that mechanism ceasing to move immediately guarantees no further damage is done by the mechanism to your hand. Sure, your hand may still be trapped, but that's better than it being, say, detached. It is then trivial to transition to a state of coast simply by cutting power to the motors. Consider the alternative - the motor coasts and may result in further damage.

To me it seems like BRAKE guarantees that the robot cannot do further damage, while potentially being a bit annoying to those damaged. COAST on the other hand may cause further damage, which is the primary thing we would like to avoid, hence I suggest using BRAKE as the safe state.

kierdavis, 2019-04-26:
Come to think of it, this choice doesn't actually matter for our current use cases. The power board will cut power to all of its outputs including any motor boards, so all motors will coast regardless of what you tell the motor board to do.

If we still want to make a (theoretical) decision on this, I think we need to come up with some realistic scenarios. As exciting as spinny sawblades of death or stabbing paraphernalia sound they're not good examples of the typical (or even worst-case) safety issues that come about during our current activities.

trickeydan, 2019-04-26:
I have come up with the following scenarios for us to consider. Feel free to add more.

## Scenario A

A robot with two tank tracks and 4 motors is travelling at maximum speed during tinker time. Due to the volunteers being busy, there are a large number of students inside the arena at this time. The robot has not been safety checked due to human error, and may contain sharp implements. The LiPo battery is close to the motor, and not particularly secure.

An ultrasound sensor malfunctions and returns a distance of zero, causing a `DivideByZeroError` in the python script controlling the robot. The python code crashes. The specific hardware of the robot means that power will not be cut from the motors unless we instruct the board to do so (e.g motor board doesn't see serial connection end, power board does not control motor board power).

The robot is heading at high velocity towards a team of innocent students.

## Scenario B

A team is working on their robot, which uses a chain and sprocket to transfer power from the motors to their axles. They are unsupervised, working after school in a rush up to a competition. Alice is testing the motors and is spinning them at full power. Meanwhile, Bob is trying to put a cable tie around a part very close to the sprocket.

The specific hardware of the robot means that power will not be cut from the motors unless we instruct the board to do so (e.g motor board doesn't see serial connection end, power board does not control motor board power).

Due to the code comparing an int to a string, the python code crashes. Bob's fingers are very close to the sprocket and could easily get caught in the spinning mechanism.

## Scenario C

A robot with two tank tracks and 4 motors is travelling at maximum speed during testing. The robot is made of flimsy plastic because it is a prototype chassis. Due to space constraints within their classroom, it is heading straight towards a solid concrete wall. The team are standing well back from the robot.

The python code crashes. The specific hardware of the robot means that power will not be cut from the motors unless we instruct the board to do so (e.g motor board doesn't see serial connection end, power board does not control motor board power).

mildlyincompetent, 2019-04-26:
Scenario A: Robot not hitting students is good, hence brake.

Scenario B: Bob's hand cannot get caught in the spinning mechanism if the mechanism is not spinning; brake.

Scenario C: Probably brake, however if the robot is flimsy aggressive breaking may result in rapid unscheduled disassembly. However, coasting in the wall is likely to result in this also.

trickeydan, 2019-06-09:
I'm going to close this in favour of #304.

closed 2019-06-09 by trickeydan