srobo/j5 #260: Should the lock port be modifiable by API authors?

trickeydan, 2019-04-23:
There could be a conflict on our port.

trickeydan, 2019-04-23:
```python
    def __new__(cls, *args, **kwargs) -> 'BaseRobot':  # type: ignore
        """Create a new instance of the class."""
        obj: BaseRobot = super().__new__(cls)

        # We have to ignore some of the types here as they are unknown.
        obj.__init__(*args, **kwargs)  # type: ignore

        obj._obtain_lock()

        return obj
```

As the lock is requested in __new__, we we need to [use a metaclass](https://stackoverflow.com/questions/14755754/pass-keyword-argument-only-to-new-and-never-further-it-to-init) to do this.

I actually think that might be slightly nicer than using `__new__` anyway.

mildlyincompetent, 2019-04-25:
I would say yes, this feels like something that can be quite important to change depending on OS (do IBM mainframes have 10k ports? I dunno, I bet they have python though so someone will try to run j5 on them at some point), or other running services.

But ultimately this comes down to what makes a good config option. In my opinion anything which is an arbitrary constant, the configuration of which may be useful to the API author, should probably be a config option (subject to common sense), and by that logic the port number should also be configurable.

kierdavis, 2019-05-31:
I agree that this would be a useful option to have.

I think the main argument against this change is that it makes situations where there are two instances of j5 on the same machine using different port numbers more likely. For example, one instance could be a background service with a non-default port number configured and the other instance is operated by a human who is quickly testing some things out at the command line (using the default port number).

I don't a strong opinion either way on this decision.