srobo/j5 #300: Brexit Strategy

trickeydan, 2019-06-09:
We need to determine how j5 will be affected by the current political environment.

We could be affected by the following :

- Shortage of python byte code due to delays in the channel tunnel
- Software may be strong and stable, but is actually weakly typed. 
- We might have to support python 2 if we end up back in the dark ages. 
- The EU might ban robots. 
- It may be harder to trade flamethrowers. 

Solutions will need to be planned and ready to implement as necessary.

antonnikitin97, 2019-06-09:
Hi Dan,

I agree we should take this into account, however I'm sure our lord and saviour Tonald Drump will come in with a mighty, tasty, zesty exclusive trade deal that will allow us to still have access to all the chlorinated chicken! 

To that end, I'll now examine each of these points.

> Shortage of python byte code due to delays in the channel tunnel

This will be a problem from the outset but due to us taking back control being a fully sovereign nation again we will be able to implement a points-based Visa system. Meaning we'll only now get skilled byte-code!

> Software may be strong and stable, but is actually weakly typed.

The typing of the programming language shouldn't have any impact on our superior fully sovereign control-taking government!

> We might have to support python 2 if we end up back in the dark ages.

At least we'll have blue passports!

> The EU might ban robots.

I don't see why this is an issue, by leaving the neo-facist, liberal, bureaucratic, anti-democratic entity known as the EU, they'll no longer have any jurisdiction over us and hence we'll be able to continue thew robotic uprising!

> It may be harder to trade flamethrowers.

Thanks to our trade deal with Freedom Land, we'll be able to continue to import a fuck-ton of flamethrowers to fend off our foes!

Glory to Arstotzka!

mildlyincompetent, 2019-08-25:
Has this issue been updated since the Borising.

Also formal petition to use the verb `commit` when referring to brexit, e.g. "the government intends to commit hard brexit in October".

trickeydan, 2019-09-28:
I am proroguing this issue.

closed 2019-09-28 by trickeydan