srobo/j5 #305: Bound Type Variable on Board.discover

trickeydan, 2019-06-09:
It is possible to use a bound type variable on `Board.discover` to more tightly type the return type.

kierdavis, 2019-06-09:
`Board.discover` hasn't existed since 229fdcf9bfcdc27dfab404a6d1e9af14835fb429. Do you mean `Backend.discover`?

trickeydan, 2019-06-09:
> `Board.discover` hasn't existed since [229fdcf](https://github.com/j5api/j5/commit/229fdcf9bfcdc27dfab404a6d1e9af14835fb429). Do you mean `Backend.discover`?

Erm, yeah I did

trickeydan, 2019-10-13:
I have attempted this and it would require to use bound Type Variables in *a lot* of places. Enough that the increase in knowledge to use the project outweighs the benefits that we get imo.

I personally would like to mark this as a `wontfix` for this reason.

trickeydan, 2019-10-13:
@kierdavis ?

trickeydan, 2019-10-13:
If it is agreed that this is `wontfix`, we should close this issue.

kierdavis, 2019-10-14:
Yeah, this sounds like a "version 2.0" task, if anything.

closed 2019-10-14 by kierdavis