srobo/j5 #315: Return value of read_gpio_pin_analogue_value is unclear

kierdavis, 2019-06-18:
It is not clearly documented what the scale of the return value of `GPIOPinInterface.read_gpio_pin_analogue_value` should be. Is it

* between 0 and 1?
* between 0 and the maximum ADC reading? (usually 255 or 1023 for an arduino)
* between 0 and the reference voltage? (usually 5 V for an arduino)

I think the last option is the most useful to the user.

https://github.com/j5api/j5/blob/master/j5/components/gpio_pin.py#L76

trickeydan, 2019-06-18:
The latter was my intention when I wrote the interface.

trickeydan, 2019-06-26:
We should clarifiy this in code, but between 0 and the reference voltage? (usually 5 V for an arduino) is what it should be

kierdavis, 2019-06-26:
Makes sense to me.

I think (not sure since it's been a while since I looked at this) that there's a few places in the code that don't follow this, so we should look through and make sure the right scale is being used consistently throughout the codebase.