srobo/j5 #344: Assert will be compiled out in optimised bytecode.

trickeydan, 2019-07-21:
This will be compiled out in optimised bytecode.

_Originally posted by @trickeydan in https://github.com/j5api/j5/pull/307_

kierdavis, 2019-07-21:
I'm not clear on what the problem is here.

trickeydan, 2019-07-21:
Leftover issues from #307

```
    def _update_digital_pin(self, identifier: int) -> None:
        assert identifier < FIRST_ANALOGUE_PIN
```

The `assert` statement will be compiled out in optimised bytecode, which could result in unknown behaviour. We should be using an `if` statement instead ideally and then throwing an exception.

There are two usages of `assert` which need removing. 

`assert` should only be used in tests for best practice.

PR #366 (2019-08-05): Remove unsightly assertions
Fixes #344.
closed 2019-08-05 by kierdavis