srobo/j5 #350: Consider use of cached_property

kierdavis, 2019-07-24:
https://pypi.org/project/cached-property/

trickeydan, 2019-07-25:
This is currently a dependency of `zoloto`, so I have no issue adding it here.

sedders123, 2019-07-25:
Where would this be useful? It may be useful in some places but right now I'm struggling to think of any methods that are computationally intensive enough to benefit from this. Could we come up with a list of expensive methods to help whoever picks this up?

kierdavis, 2019-07-25:
[`Coordinate.cylindrical` and `Coordinate.spherical`](https://github.com/j5api/j5/blob/master/j5/vision/coordinates.py#L46) are already using explicit caching. Switching to `cached_property` would make the code a bit neater here.

trickeydan, 2019-07-25:
> [`Coordinate.cylindrical` and `Coordinate.spherical`](https://github.com/j5api/j5/blob/master/j5/vision/coordinates.py#L46) are already using explicit caching. Switching to `cached_property` would make the code a bit neater here.

I agree.

> It could also be used for `Backend.firmware_version`, though I think most backends retrieve the firmware version during initialisation in order to check it is a supported version.

Thus the property is never fetched twice, the firmware version won't randomly change mid-execution.

kierdavis, 2019-07-25:
> > It could also be used for `Backend.firmware_version`, though I think most backends retrieve the firmware version during initialisation in order to check it is a supported version.
> 
> Thus the property is never fetched twice, the firmware version won't randomly change mid-execution.

Sorry, I completely bodged what I was trying to say there I think. That's correct. What I was aiming to say was that we could move the retrieval of the firmware version into the `firmware_version` cached property itself, and then access it from the constructor if needed.

trickeydan, 2019-07-25:
> Sorry, I completely bodged what I was trying to say there I think. That's correct. What I was aiming to say was that we could move the retrieval of the firmware version into the `firmware_version` [...]

I don't think we're disagreeing :P

trickeydan, 2020-01-02:
Moved to 1.0.0.

PR #530 (2020-02-03): Use cached_property in Coordinate class
Fixes #350
closed 2020-02-03 by trickeydan