srobo/j5 #363: opencv-python-contrib should be an optional dependency

kierdavis, 2019-08-03:
Ideally it should only be installed when the `zoloto-vision` extra is selected.

kierdavis, 2019-09-01:
Unfortunately this cannot be included in the `zoloto-vision` extra, because it is a dev dependency. [Apparently this is by design.](https://github.com/sdispater/poetry/pull/606#issuecomment-437943927)

It also can't be made a floating optional dependency, because there's no way to install an optional dependency other than by referencing it from an extra. [This is also by design.](https://github.com/sdispater/poetry/issues/426#issuecomment-421022108)

What is this package needed for again? It's annoying to constantly have an unstaged change in my working copy (I can't install this package because of #352).

trickeydan, 2019-09-01:
Ok, I have a suggestion. 

As a trial for #97 we could move zoloto to it's own package. This will require a reasonable amount of work, but we need to put it in anyway.

As a temporary solution, I'm ok removing this as a dev package. It's broken on some systems evidently.

closed 2019-11-01 by trickeydan