srobo/j5 #371: Investigate use of Pyre

trickeydan, 2019-08-07:
Kajetan has rightly pointed out that mypy is pretty awful.

Let us be enlighted by Ocaml

trickeydan, 2019-09-10:
```
pyre --source-directory j5 --verbose check
 ƛ Setting up a `.pyre_configuration` with `pyre init` may reduce overhead.
 ƛ Processed 30 of 57 sources [676.6s]
```

It's been stuck like this eating 100% of my CPU for 10 minutes.

trickeydan, 2019-10-19:
I've tried this again with the same result.

I'm going to call this a won't fix.

closed 2019-10-19 by trickeydan

kierdavis, 2019-10-21:
I get different output, but it still sits there pegging 100% CPU for multiple minutes.

Even if it does eventually finish, I'd deem this a major workflow breaker.

trickeydan, 2019-10-21:
I'm up for reopening this issue in, say 6 months, to see if it starts working.