srobo/j5 #388: Consider alternatives to poetry

trickeydan, 2019-08-22:
The pypoetry project are not fixing breaking bugs. We should look for an alternative. Perhaps [Flit](https://flit.readthedocs.io)?

https://github.com/sdispater/poetry/issues/536

trickeydan, 2019-08-22:
Also https://github.com/ofek/hatch

mildlyincompetent, 2019-08-25:
Note that this issue is currently quite important, as our docs simply list the tools you need to install.

For all of these tools *except* poetry, simply going to the tool's website and following their guidance will get you a working install in the vast majority of cases (unless you have a really funky system, in which case you should know what to do).

Currently, following the recommended method of installing poetry (which by the way is curl piped into python, which is pretty nasty already) will give you a broken install on standard Ubuntu (!!) and any other system which follows the Python standard laid out in PEP 394 whereby `python` is Python 2 and `python3` is Python 3. Poetry helpfully doesn't bother mentioning this anywhere except in the issue above, and no clear and obvious fix is available due to poetry's obscure method of installation.

We are in a position where we either need to document how to get poetry working on PEP 394 compliant systems (which, in my humble opinion, shouldn't be our problem), or we need to find an alternative to poetry where the default behaviour on a standards-compliant system is not horrifically broken.

trickeydan, 2019-08-25:
I did do a lot of research into this, and it is unfortunate the `poetry` is the only tool with the feature-set that we need.

To clarify this further, we should start enumerating features that we need.

A start:

- Easy to Install for new contributors
- Installs a locked version of development dependencies
- Ideally uses `pyproject.toml`
- Dependency Resolution
- Project build and publishing
- Supports Python 3.6+

mildlyincompetent, 2019-08-25:
I'm inclined to agree, but the reason I think we should keep looking for alternatives is precisely that poetry catastrophically fails at point number 1: "Easy to Install for new contributors". I still haven't managed to get it working outside of docker on Ubuntu 18.04.

trickeydan, 2019-08-30:
`Poetry will pick up the currently activated Python version to create or use the virtualenvs.`

Apparently using `pyenv` is a solution. If this is the case and we stick with poetry, we should definitely document that.

sjuxax, 2019-09-10:
Anything that overrides the parent process's PATH to make `python` point to the intended Python version will "work", but this will almost always be iffy and require local configuration. poetry really needs to understand that invoking plain old `python` is always going to cause problems for people trying to use Python 3 (if you're using Python 2.x, you may not notice). The issue essentially makes poetry incompatible with almost all Python 3 distributions.

trickeydan, 2019-10-14:
After some discussion, the poetry maintainers have shown that they will fix this before the `1.0` release. See https://github.com/sdispater/poetry/issues/536#issuecomment-541625301 for details.

I think that as I've been able to find a suitable alternative, we can consider this issue done. There does not seem to be another option with the featureset of poetry.

closed 2019-10-14 by trickeydan