srobo/j5 #394: Investigate strange exception in MB tests

trickeydan, 2019-08-26:
```
tests/backends/hardware/sr/v4/test_motor_board.py .......Exception ignored in: <function SRV4MotorBoardHardwareBackend.__del__ at 0x7fa76805a8c0>
Traceback (most recent call last):
  File "/home/dan/Code/sro/j5/j5/backends/hardware/sr/v4/motor_board.py", line 94, in __del__
    self.set_motor_state(i, MotorSpecialState.BRAKE)
  File "/home/dan/Code/sro/j5/j5/backends/backend.py", line 32, in new_method
    retval = old_method(*args, **kwargs)
  File "/home/dan/Code/sro/j5/j5/backends/hardware/sr/v4/motor_board.py", line 153, in set_motor_state
    self.send_command(CMD_MOTOR[identifier], value)
  File "/home/dan/Code/sro/j5/j5/backends/hardware/j5/serial.py", line 34, in catch_exceptions
    return func(*args, **kwargs)
  File "/home/dan/Code/sro/j5/j5/backends/hardware/sr/v4/motor_board.py", line 108, in send_command
    "Mismatch in command bytes written to serial interface.",
j5.backends.backend.CommunicationError: Mismatch in command bytes written to serial interface.
```

kierdavis, 2019-08-26:
Huh, I've never noticed this error message in the source code before, let alone at runtime. Apparently it occurs when we try to write data to the serial port but not all bytes get written (maybe because the serial write buffer is full?). Isn't this normally handled by trying to write the remaining data again?

trickeydan, 2019-10-22:
This can be replicated by running `pytest -s tests`

trickeydan, 2019-10-22:
I believe this bug is related to the crashes that don't stop the motors turning :o

PR #446 (2019-11-07): Use good serial driver for graceful shutdown in Bad Write tests.
Fixes #394

The bad serial driver was being used to destruct the Motor Driver which
was causing an obscure issue.

This took a while to debug. The flow of code was not obvious. I just
looked at the constructor and assumed that it was using MotorSerial
throughout
commit 6d27ebf9 (2019-11-07): Use good serial driver for graceful shutdown in Bad Write tests.

Fixes #394

The bad serial driver was being used to destruct the Motor Driver which
was causing an obscure issue.

This took a while to debug. The flow of code was not obvious. I just
looked at the constructor and assumed that it was using MotorSerial
throughout
closed 2019-11-07 by kierdavis