srobo/j5 #396: Technical Debt around Protocol

trickeydan, 2019-08-26:
```
if TYPE_CHECKING: 

    from typing_extensions import Protocol  # pragma: nocover 

else: 

    class Protocol: 

        """Dummy class since typing_extensions is not available at runtime.""" 

 

        pass 
```

We should see if there is a better way to do this.

`j5/backends/j5/serial.py`

kierdavis, 2019-08-26:
We could simply add `typing_extensions` as a normal dependency. Currently we rely on it being included in mypy, which is why it's only available when type checking.

Also note:

> The Protocol base class is provided in the typing_extensions package for Python 2.7 and 3.4-3.7. Starting with Python 3.8, Protocol is included in the typing module.
>
> (https://mypy.readthedocs.io/en/latest/protocols.html#simple-user-defined-protocols)

trickeydan, 2019-08-26:
I'm happy with this being added as a 3.6 and 3.7 only dependency.

kierdavis, 2019-08-26:
The poetry documentation provides no information how to make a dependency only apply to particular python versions. Is this possible?

trickeydan, 2019-08-26:
> The poetry documentation provides no information how to make a dependency only apply to particular python versions. Is this possible?

Unsurprising. My opinion of poetry is low. Relevant: #388 

https://github.com/RealOrangeOne/zoloto/blob/e298adb5c108e3dabde3c10db461e62fb08889c9/pyproject.toml#L55

kierdavis, 2019-08-29:
```
root@7c80b02d344f:~/j5# python --version
Python 3.8.0b3
root@7c80b02d344f:~/j5# grep typing-extensions pyproject.toml
typing-extensions = {version = "^3.7.4", python = "<3.8"}
root@7c80b02d344f:~/j5# poetry lock
Updating dependencies
Resolving dependencies... (0.9s)
root@7c80b02d344f:~/j5# poetry install --dry-run | grep typing-extensions
  - Installing typing-extensions (3.7.4)
```

Any idea what I'm doing wrong here?

trickeydan, 2019-08-30:
> Any idea what I'm doing wrong here?

Upon some investigation, it doesn't look like this works at all. I'm particularly curious about why it works for the zoloto tests.

There doesn't seem to be a feature like this in `pip`, `pipenv` or `flit` either.

Opened https://github.com/RealOrangeOne/zoloto/issues/67 to track on Zoloto side

trickeydan, 2019-08-30:
Apparently it does work on Zoloto, and you can generate it with the cli: https://github.com/RealOrangeOne/zoloto/issues/67#issuecomment-526624182

```
╭─dan@dgt-laptop ~/oof-logos
╰─$ poetry add -h
Usage:
  add [options] [--] <name> (<name>)...

Arguments:
  name                            Packages to add.

Options:
  -D, --dev                       Add package as development dependency.
      --git=GIT                   The url of the Git repository.
      --path=PATH                 The path to a dependency.
  -E, --extras=EXTRAS             Extras to activate for the dependency. (multiple values allowed)
      --optional                  Add as an optional dependency.
      --python=PYTHON             Python version( for which the dependencies must be installed.
      --platform=PLATFORM         Platforms for which the dependencies must be installed.
      --allow-prereleases         Accept prereleases.
      --dry-run                   Outputs the operations but will not execute anything (implicitly enables --verbose).
  -h, --help                      Display this help message
  -q, --quiet                     Do not output any message
  -V, --version                   Display this application version
      --ansi                      Force ANSI output
      --no-ansi                   Disable ANSI output
  -n, --no-interaction            Do not ask any interactive question
  -v|vv|vvv, --verbose[=VERBOSE]  Increase the verbosity of messages: 1 for normal output, 2 for more verbose output and 3 for debug
```

This seems to be in the cli.

trickeydan, 2019-08-30:
```
╭─dan@dgt-laptop /tmp/shitty-python/poetry-attempt-1
╰─$ poetry --version
Poetry 0.12.17
╭─dan@dgt-laptop /tmp/shitty-python/poetry-attempt-1
╰─$ python --version
Python 3.7.4
╭─dan@dgt-laptop /tmp/shitty-python/poetry-attempt-1
╰─$ poetry add dataclasses --dry-run --python="<3.7"
Using version ^0.6.0 for dataclasses

Updating dependencies
Resolving dependencies... (0.1s)

Nothing to install or update

╭─dan@dgt-laptop /tmp/shitty-python/poetry-attempt-1
╰─$ poetry add typing_extensions --dry-run --python="<3.8"
Using version ^3.7 for typing_extensions

Updating dependencies
Resolving dependencies... (1.5s)

Package operations: 2 installs, 0 updates, 0 removals

  - Installing typing (3.7.4.1)
  - Installing typing-extensions (3.7.4)
```

trickeydan, 2019-08-30:
@kierdavis What version of poetry are you using?

kierdavis, 2019-09-01:
Oh, you are correct, this all works as expected. I forgot that `mypy` has a dependency on `typing-extensions`, so it will be installed regardless of the python version unless you specify `--no-dev`.

```
root@2e4bec29c282:~/j5# python --version
Python 3.7.4
root@2e4bec29c282:~/j5# poetry install --no-dev --dry-run | grep typing-extensions
  - Installing typing-extensions (3.7.4)
```

```
root@76dc57d501d7:~/j5# python --version
Python 3.8.0b3
root@76dc57d501d7:~/j5# poetry install --no-dev --dry-run | grep typing-extensions
  - Not removing typing-extensions (3.7.4) Not currently installed
```

```diff
diff --git a/pyproject.toml b/pyproject.toml
index d03ee89..6f2282e 100644
--- a/pyproject.toml
+++ b/pyproject.toml
@@ -22,6 +22,7 @@ pyusb = "^1.0"
 pyserial = "^3.4"
 pyquaternion = "^0.9.5"
 zoloto = {version = "^0.3.0", optional = true}
+typing-extensions = {version = "^3.7",python = "<3.8"}
 
 [tool.poetry.dev-dependencies]
 flake8 = "^3.7"
```

I'll make a PR now.

kierdavis, 2019-09-01:
Ugh, I can't do

```python
try:
    from typing import Protocol
except ImportError:
    from typing_extensions import Protocol
```

because mypy considers `typing.Protocol` and `typing_extensions.Protocol` to be of different and incompatible types:

```
j5/backends/hardware/j5/serial.py:15: error: Incompatible import of "Protocol" (imported name has type "typing_extensions._SpecialForm", local name has type "typing._SpecialForm")
```

Seeing as `Protocol` is unlikely to be removed from `typing_extensions` even after it gets merged into `typing`, we might as well always import it from the former.

PR #400 (2019-09-01): Remove typing extensions import check
Fixes #396.
commit 4ea2337a (2019-09-01): Remove import check for typing-extensions

As of fde5b7ed0959719d2a7ac9faf2169720df6c9078 it is now always available.

Fixes #396.
closed 2019-09-01 by kierdavis