srobo/j5 #402: Separate Zoloto into it's own package

trickeydan, 2019-09-01:
Ok, I have a suggestion. 

As a trial for #97 we could move zoloto to it's own package. This will require a reasonable amount of work, but we need to put it in anyway.

As a temporary solution, I'm ok removing this as a dev package. It's broken on some systems evidently.

_Originally posted by @trickeydan in https://github.com/j5api/j5/issues/363#issuecomment-526943265_

kierdavis, 2019-09-03:
I suggest we start by pulling out all the vision stuff together (which is the minimum needed for #363), and then work on decoupling the zoloto-specific stuff from the non-zoloto-specific stuff.

trickeydan, 2019-09-03:
I'm currently working on a PR to enable plugin functionality, I'll submit that soon and then we can begin the package separation.

I think it would be good to use a monorepo for this. @mildlyincompetent is likely to be better at explaining the advantages than me.

sedders123, 2021-06-27:
I'm +1 on this idea, also have no objections to monorepo.

PR #718 (2021-07-04): Remove vision support
## Checklist

- [x] I have updated the relevant documentation
- [x] I have added the `semver-major`, `semver-minor` or `semver-patch` label as appropriate
- [x] I would like multiple reviewers to approve my code before merging

## Why do you want to make these changes?

Following discussions in #715, I believe the best way forward is to move vision code into a `j5-zoloto` module.

I intend to directly copy all of the removed code to a separate repo, and will attempt to preserve history if possible.

De-coupling zoloto and j5 will allow us to resolve issues with duplicated [... cut]
closed 2021-07-04 by trickeydan