srobo/j5 #403: Explore auto-releasing

trickeydan, 2019-09-01:
- Annotated, Signed Git Tags.

I'll have a look at this at some point, but I want to concentrate on some other things for now, and we do currently have a working release process.

trickeydan, 2019-09-16:
I'm unassigning myself as I think there are more important things to do.

trickeydan, 2019-10-29:
GitHub actions is a much nicer way to do this.

sedders123, 2019-10-29:
I have implemented a [GitHub Action](https://github.com/sedders123/phial/blob/develop/.github/workflows/cd.yml) to deploy to PyPi for one of my personal projects. While it probably can't be copied exactly it may serve as a decent jumping off point.

trickeydan, 2021-08-08:
> I have implemented a [GitHub Action](https://github.com/sedders123/phial/blob/develop/.github/workflows/cd.yml) to deploy to PyPI for one of my personal projects. While it probably can't be copied exactly it may serve as a decent jumping off point.

I really like the `verify` stage of this.

PR #727 (2021-08-08): Publish to PyPI on Release
## Checklist

- [x] I have updated the relevant documentation
- [ ] I have added the `semver-major`, `semver-minor` or `semver-patch` label as appropriate
- [ ] I would like multiple reviewers to approve my code before merging

## Why do you want to make these changes?

Fixes #403 

## Which parts of the codebase do your changes affect?

None.

## How do your changes affect student or volunteer experience?

No more manual releases, less bus factor

## Are your changes related to any existing issues or PRs? How so?

Fixes #403
closed 2021-08-08 by trickeydan