srobo/j5 #436: AttributeError: 'SRV4MotorBoardHardwareBackend' object has no attribute '_state'

trickeydan, 2019-10-16:
```python
from sbot import *
from time import *

r = Robot()

while True:
    print(r.motor_boards) 
```

```
Script started on 2019-08-14 18:19:58+01:00 [<not executed on terminal>]
sbot.robot - SourceBots API v0.7.0
Traceback (most recent call last):
  File "main.py", line 5, in <module>
    r = Robot()
  File "/usr/local/lib/python3.7/dist-packages/sbot/robot.py", line 64, in __init__
    self._init_auxilliary_boards()
  File "/usr/local/lib/python3.7/dist-packages/sbot/robot.py", line 81, in _init_auxilliary_boards
    HardwareEnvironment.get_backend(MotorBoard),
  File "/usr/lib/python3.7/typing.py", line 670, in __call__
    result = self.__origin__(*args, **kwargs)
  File "/usr/local/lib/python3.7/dist-packages/j5/boards/board.py", line 125, in __init__
    self.update_boards()
  File "/usr/local/lib/python3.7/dist-packages/j5/boards/board.py", line 130, in update_boards
    discovered_boards = self._backend_class.discover()
  File "/usr/local/lib/python3.7/dist-packages/j5/backends/hardware/sr/v4/motor_board.py", line 59, in discover
    SRV4MotorBoardHardwareBackend(port.device, serial_class),
  File "/usr/local/lib/python3.7/dist-packages/j5/backends/hardware/j5/serial.py", line 34, in catch_exceptions
    return func(*args, **kwargs)
  File "/usr/local/lib/python3.7/dist-packages/j5/backends/hardware/sr/v4/motor_board.py", line 70, in __init__
    baud=1000000,
  File "/usr/local/lib/python3.7/dist-packages/j5/backends/hardware/j5/serial.py", line 34, in catch_exceptions
    return func(*args, **kwargs)
  File "/usr/local/lib/python3.7/dist-packages/j5/backends/hardware/j5/serial.py", line 90, in __init__
    timeout=timeout_secs,
  File "/usr/local/lib/python3.7/dist-packages/serial/serialutil.py", line 240, in __init__
    self.open()
  File "/usr/local/lib/python3.7/dist-packages/serial/serialposix.py", line 288, in open
    self._update_rts_state()
  File "/usr/local/lib/python3.7/dist-packages/serial/serialposix.py", line 627, in _update_rts_state
    fcntl.ioctl(self.fd, TIOCMBIS, TIOCM_RTS_str)
OSError: [Errno 5] Input/output error
Exception ignored in: <function SRV4MotorBoardHardwareBackend.__del__ at 0x74aaffa8>
Traceback (most recent call last):
  File "/usr/local/lib/python3.7/dist-packages/j5/backends/hardware/sr/v4/motor_board.py", line 93, in __del__
    for i, val in enumerate(self._state):
AttributeError: 'SRV4MotorBoardHardwareBackend' object has no attribute '_state'

Script done on 2019-08-14 18:20:01+01:00 [COMMAND_EXIT_CODE="1"]

Your code was unsuccessful.
```

trickeydan, 2019-10-16:
I have been unable to replicate this again, but I think the crash condition is interesting. It crashes in the setup of the serial device

kierdavis, 2019-10-17:
I've never seen that `OSError` before - perhaps the serial port disappeared halfway through initialisation?

The `AttributeError` in `__del__`  is because the code crashed [here](https://github.com/j5api/j5/blob/4641ca4fe579d29be1659de2349cb298e5feb2fd/j5/backends/hardware/sr/v4/motor_board.py#L65), which is before `self._state` gets assigned [here](https://github.com/j5api/j5/blob/4641ca4fe579d29be1659de2349cb298e5feb2fd/j5/backends/hardware/sr/v4/motor_board.py#L65). I suggest we wrap the stuff in `__del__` that uses `self._state` with a `hasattr` check.

PR #447 (2019-10-23): Ensure that the Motor Board shuts down safely if the serial hasn't initialised
Fixes #436
commit cfe5874c (2019-10-23): Ensure that the Motor Board shuts down safely if the serial hasn't
initialised

Fixes #436
closed 2019-10-23 by kierdavis