srobo/j5 #444: Remove references to Environment where it is not used

trickeydan, 2019-10-22:
When we changed the use of environments, they were left sprinkled in a few places uselessly.

For example `tests/test_stack.py`.

We should remove these.

udhayacommits, 2019-10-23:
Hello @trickeydan ,
I would like to take this up.

trickeydan, 2019-10-23:
Hi @udhayacommits,

That's great! Thanks for wanting to contribute to our project.

This issue arises out of #406 , where we changed the meaning of the `Environment` class. Prior to this PR, we provided predefined `Environment` instances which already contained `Backend` classes registered with them. This is not obvious from the old code as it used metaclasses for implicit registration.

There is at least one case (i.e in `tests/test_stack.py`) where we are still using an `Environment` in the old way. There may be more. Probably mostly in the `tests`. I'd recommend searching through the codebase for usages of `Environment`.

As this is your first time contributing, you may find our [Getting Started with j5 Development Guide](https://j5.readthedocs.io/en/master/development/getting-started.html) useful.

Please feel free to ask me or @kierdavis if you have any questions. :)

udhayacommits, 2019-10-23:
Hi @trickeydan, I have gone through the codebase for occurences of Environment. But I am unsure of how to proceed with the changes and the impact it will cause.
For example, under tests/backends/test_backend.py I can find several usages for `Environment`. Are these to be removed completely

trickeydan, 2019-12-21:
Fixed in #450

closed 2019-12-21 by trickeydan