srobo/j5 #468: Orientation class doesn't convert from cartesian properly

tpoomlmly, 2019-11-23:
Specifically in the `from_cartesian` method. Currently, the method creates a new quaternion that represents a rotation by `x` about the x axis. It then creates a quaternion for a rotation by `y` radians about the y axis and rotates the first quaternion by the second one. It does this again with `z`. Each subsequent rotation in a different axis messes with the previous rotations, like twisting a Rubik's cube. 
Rotating in the order z-x-y produces a different result to rotating in the order x-y-z since quaternion rotation is done by multiplication, and quaternion multiplication is non-commutative.

To fix this, all the rotations need to be performed at the same time, by using the yaw, pitch and roll to calculate the four components of the quaternion. `pyquaternions` doesn't do this for us.

Also, the method's name and docstring are misleading. It has nothing to do with René Descartes or his coordinate system.

tpoomlmly, 2019-11-23:
- [ ] Fix the logic
- [ ] Fix the name
- [ ] Fix the docstring

trickeydan, 2019-12-06:
Tyler suggested that we use the `rvec` rotation vector that ArUco provides and turn it directly into a quaternion.

trickeydan, 2020-01-16:
Tyler's suggestion was submitted as an upstream PR:https://github.com/RealOrangeOne/zoloto/pull/127

We should be able to directly use the quaternion that this exposes.

trickeydan, 2020-01-19:
Zoloto released 0.5.0

PR #505 (2020-01-23): Update to Zoloto v0.5.0
- Fixes Orientation; Closes #468 
- Adds support for discovery of cameras.
  - Multiple Camera Support
  - Adds cross-platform support.
- Removes orientation support from the ZolotoCameraBoardConsoleBackend.
  - It would require complex maths that is out of scope imo.
- Use `process_frame_eager`; Fixes #347
closed 2020-01-23 by trickeydan