srobo/j5 #492: Source bundle on PyPi does not include tests

mildlyincompetent, 2020-01-08:
The j5 source bundle on PyPi does not include tests. This means that build-from-source distributions like NixOS which use PyPi as the python package source of truth cannot easily build and test the package.

This could be resolved by including the tests in the source bundle, but not in the wheels or other prebuilt packages.

trickeydan, 2020-01-09:
@mildlyincompetent Are you willing to take on this issue?

Possibly useful things:

- https://python-poetry.org/docs/pyproject/#include-and-exclude
- https://python-poetry.org/docs/pyproject/#packages

As this is a packaging change, `poetry build` is included as a thing that we will need to test.

trickeydan, 2021-07-17:
I don't think we need this as we are not actively developing on NixOS anymore. If the requirement comes up again in the future, I'd be happy to re-open this issue.

closed 2021-07-17 by trickeydan