srobo/j5 #571: Switch to external PR workflow only

trickeydan, 2020-05-29:
Would it be better to have everyone follow the same (less privileged) workflow? This could help us surface issues in things like CI that may not appear until someone outside the organisation try to contribute which would impeded us merging in there changes without some potentially unnecessary back and forth.

_Originally posted by @sedders123 in https://github.com/j5api/j5/pull/539_

We should test the external PR as part of this issue, and update the documenation accordingly.

trickeydan, 2020-07-28:
Reopening until #609 gets closed and the branch deleted. This is the last branch on the main repo.

After that is merged, we should email the mailing list and restrict writes.

sedders123, 2020-07-29:
#609 looks to be merged, is this unblocked now?

trickeydan, 2020-07-30:
To Do:

- [ ] Check that the docs reflect the new workflow
- [ ] Email the j5 mailing list to ensure that contributors know
- [ ] Lock down pushing of branches if possible

PR #573 (2020-05-31): Ensure everyone uses the less privileged workflow
This PR partially exists so that I can test the CI

## Checklist

- [x] I have updated the relevant documentation
- [x] I have added the `semver-major`, `semver-minor` or `semver-patch` label as appropriate
- [ ] I would like multiple reviewers to approve my code before merging

## Why do you want to make these changes?

All contributors should use the less-privileged workflow to ensure that our CI behaves appropriately.

## Which parts of the codebase do your changes affect?

None

## How do your changes affect student or volunteer experience? [... cut]
commit 5847c08b (2020-05-31): Everyone should use the less-privileged workflow

Fixes #571