srobo/j5 #615: Add Development documentation for Windows

trickeydan, 2020-07-28:
We should support this.

- [ ] Use of powershell
- [ ] Make?
- [ ] Zadig (Need to work out how, maybe ask Tyler?)

sedders123, 2020-07-29:
Happy to help out with this if needed. I'm thinking we should try to setup j5 in a [Windows Sandbox](https://techcommunity.microsoft.com/t5/windows-kernel-internals/windows-sandbox/ba-p/301849) so that we can document all dependencies and not miss anything. With regards to Zadig, I'm no USB expert but I'm happy to download and give it a poke. I imagine my initial pass of these docs would be purely around the project setup. 

I won't assign myself to this until I have capacity to actually work on it (hopefully sometime next week), happy for someone else to pickup in the meantime though!

trickeydan, 2023-01-14:
Using libusb-package like april_vision does might help with this.