srobo/j5 #715: Should we de-scope vision?

trickeydan, 2021-06-19:
I'd like to propose that we de-scope vision from `j5` and remove all related code. We can point developers in the direction of Zoloto and other similar libraries.

**Pros**
- We are currently duplicating half of the human friendly API from Zoloto and this is a pain to maintain.
- Less effort to implement vision on APIs, especially as it's usually a custom use case
- We only support one vision system anyway, and it's currently broken!
- We can remove a lot of untested code
- No dependencies on OpenCV or pyquaternion, or numpy

**Cons**
- Less features

RealOrangeOne, 2021-06-19:
Disclaimer: As the maintainer of [Zoloto](https://github.com/RealOrangeOne/zoloto), my opinion may be bias, but equally educated.

I like the idea of removing the duplication of code, there's no need to have the duplication and wrapper just for the sake of a generic and composable API. It also means we can focus developer time into creating features and improving the platform, rather than creating shims for things which already exist.

With that said, there is a way around this which may be better: A strict dependency on zoloto (optional). Therefore just working under the assumption that "j5 vision uses zoloto and only zoloto". Therefore rather than specific j5 APIs, it's just exposing Zoloto constructs, with perhaps some wrappers to support j5s different backends (these would likely be very simple). Obviously there are lots of cases where vision isn't required, and OpenCV can be a big dependency, so probably want it as an optional dependency (`j5[vision]`).

Zoloto has specifically been designed to be generic, widely useful, and highly configurable, so it should fit nicely with the feature and composability requirements and ethos of j5.

sedders123, 2021-06-19:
My gut says descope it, but gonna spew some thoughts below and see where I end up.

<hr>

> * We are currently duplicating half of the human friendly API from Zoloto and this is a pain to maintain.

Our own Vision API does allow us to shim around Zoloto breaking changes and give us the ability to switch vision provider if we wanted to.
> * Less effort to implement vision on APIs, especially as it's usually a custom use case

I think this means that consumers of `j5` would have more fine grained control over how there visions system works as they would implement it themselves. The flipside of that is that if consumers want a vision system they have to implement their own.
> * We only support one vision system anyway, and it's currently broken!

We only support one power board at the moment. We could fix our current vision system.
> * We can remove a lot of untested code

Or we could test it
> * No dependencies on OpenCV or pyquaternion, or numpy

Fewer dependencies would be nice, but if we're expecting consumers to roll their own vision they'll likely end up with these anyway.

<hr>

If j5's aim is to create consistent APIs a facade in front of different vision providers would be needed. This facade would either have to only contain the minimum attributes that all vision providers can offer, or allow for attributes and methods to be null or throw not implemented exceptions. Limiting ourselves to a subset of features feels limiting and letting end users think loads of super fancy methods exist only for errors to appear at runtime isn't great. This is what we currently have for all our components but given the broadness of Computer Vision, vs say a Motor Board, it feels much harder to design a simple common API suitable for all use cases.

Having a `j5[vision]` which was a thin wrapper around Zoloto would certainly save us some de duplication and would still provide plenty of configuration to end users. It would also let consumers have an 'out of the box' vision system. However we would be tying ourselves to Zoloto and its public API and given Zoloto is still pre 1.0.0, and as such has an unstable API, this feels a little risky. (I'm aware we are also in initial development but I'd hope we're getting pretty close to 1.0.0 given the success of sbot and the current development of sr.robot3). Also if another vision provider came along we'd like to switch to that would be a breaking change. Maybe a best of both worlds is a separate `j5-zoloto` package giving us room to switch vision providers in the future and the ability to support multiple major versions of zoloto if necessary.

Apologies for the wall of text. tl;dr; Maybe we should have a separate `j5-zoloto` package.

sedders123, 2021-06-19:
(Also should we think about enabling [GitHub Discussions](https://docs.github.com/en/discussions) for future discussions?)

trickeydan, 2021-06-27:
I'd be in favour of a separate `j5-zoloto` package, much like how was suggested in #402 and #97. It could provide Zoloto-specific components and backends.

We never intended to build this in permanently, it was always a stopgap during development.

I'd really like to see j5 go 1.0 in the next couple of months and I don't want that to rely on the stability of external software.

What do you both think?

trickeydan, 2021-06-27:
We could definitely consider bring some of the vision components back into `j5` later once they are more stable e.g `Marker` and `MarkerCamera` component classes. iirc these were planned to be factored out into `j5-vision` as part of #97

RealOrangeOne, 2021-06-27:
#97 definitely sounds along the right lines of this, and something this follows the spirit of. That said it is slightly different. `j5.sr` would be a set of _implementations_ of existing j5 interfaces. `j5-vision` would be entirely separate, and potentially a dependency of `j5.sr`.

I think we'd need to understand what would live in `j5`, `j5.sr` (vision related) and `j5-zoloto`.

trickeydan, 2021-06-27:
> I think we'd need to understand what would live in `j5`, `j5.sr` (vision related) and `j5-zoloto`. 

I'd suggest just pulling out Zoloto and vision code for now and leaving the SR drivers in j5 core. I think that pulling out sr drivers would have diminishing returns. Python monorepos are hard to maintain and would hinder taking j5 stable (which we need before kick-start really).

Proposal:

We pull out all vision related code, including Zoloto into `j5-zoloto`. This would include components, boards and backends for using zoloto/libkoki style marker vision.

I suspect that CV is not generalisable enough for a j5-vision package as that use case is fairly specific.

RealOrangeOne, 2021-06-28:
That approach sounds good to me. The package itself should be pretty small, making testing and maintaining it very simple. Exactly what would need to live in there and how easy it would be to integrate with existing j5 _things_ I don't know, but I don't doubt the design of j5 makes that reasonably simple.

Potentially, some of the primitives implemented in https://github.com/RealOrangeOne/zoloto/issues/279 could be useful here.

trickeydan, 2021-07-17:
Vision was de-scoped in #718.

closed 2021-07-17 by trickeydan