srobo/j5 #758: Ensure arduino pins are set as outputs when changing mode

RealOrangeOne, 2021-11-28:
Currently, when setting a pin as `OUTPUT`, the arduino never gets the message. For all other pins it works as expected.

The cause of this is in:

https://github.com/j5api/j5/blob/4517772684615516e73a09edad751df6ed17bb44/j5/backends/hardware/sr/v4/ruggeduino.py#L103-L107

Note how only the state is set, not the pin mode, which causes weird things to happen. This could be patched with firmware, but that's a lot of work, and arguably fixing it here is the correct course of action anyway.

WillB97, 2021-12-01:
There are kind of 2 issues here:
1. `_update_digital_pin` doesn't set the output mode
2. Setting the pin mode and setting the output state should be separate functions as we don't want to repeatedly set the output mode

trickeydan, 2021-12-02:
Issue confirmed.

commit 2a9bbc64 (2021-12-02): backends/ruggeduino: Set the pin mode when writing digital outputs

This patch does set the pin mode on every output, which is identical in behaviour to how the SB Arduino backend also works.

As the _update_digital_pins function does not know the previous state of the pin, we cannot know when we need to change it,
therefore we set it every single time.

Additionally, when the pin mode is set to digital output, the pin will also receive a write call, as the state in the j5 code
will already contain a state for the pin (LOW by default). This ensures that the state of the pin, matches the state  [... cut]
commit 75ff8e27 (2021-12-02): backends/ruggeduino: Set the pin mode when writing digital outputs

This patch does set the pin mode on every output, which is identical in behaviour to how the SB Arduino backend also works.

As the _update_digital_pins function does not know the previous state of the pin, we cannot know when we need to change it,
therefore we set it every single time.

Additionally, when the pin mode is set to digital output, the pin will also receive a write call, as the state in the j5 code
will already contain a state for the pin (LOW by default). This ensures that the state of the pin, matches the state  [... cut]
closed 2021-12-07 by trickeydan