srobo/j5 #839: Replace `CommunicationError` with more specific error types

raccube, 2022-12-03:
Using `ComminicationError` everywhere isn't particularly detailed and could be improved by creating specific error types for each occurrance. This would potentially allow for users to implement better error handling if a method returns multiple errors.

ShaziahGafur, 2022-12-04:
Hi @raccube, please assign this to me

trickeydan, 2022-12-04:
> Hi @raccube, I can work on this. Please assign it to me

Hi @ShaziahGafur, that sounds great! I've assigned this issue to you. If you have any questions please just ask :)

Please could you make sure that any new exception classes inherit from `CommunicationError` so that the changes aren't breaking? Thanks!